pion/mediadevices #46: Add support to ToI420 other than YCbCr

lherman-cs, 2020-02-14:
As of now, our video encoders, openh264 and vpx, only support I420 format. Therefore, we have `ToI420` converter that helps to handle any kind of image format.

However, `ToI420` can only handle YCbCr images at this moment. So, the motivation for adding more image formats would be,

1. Allow more flexibility for the input
2. Since we allow users to transform the video through `VideoTransform`, having a flexible input will also open a door for people to use a library such as https://godoc.org/github.com/disintegration/imaging, which relies highly on `NRGBA`

at-wat, 2020-02-15:
This package seems nice: https://pkg.go.dev/golang.org/x/image/draw
https://blog.golang.org/go-imagedraw-package#TOC_6.
Scaling is also available in this package.

lherman-cs, 2020-02-15:
That looks really cool. The package seems to be pretty flexible, I like it.

closed 2020-02-17 by lherman-cs