pion/mediadevices #61: Pass also requested properties to Video/AudioRecord

at-wat, 2020-02-16:
Only selected prop is passed to `Video/AudioRecord` at now.

For example on screen capture driver, FrameRate is not discrete.
It would be nice to pass the selected prop and also the requested prop to read such parameters.

lherman-cs, 2020-02-18:
I definitely forgot about handling non-discrete values in the design. At first, I figured that it was nice to have this design because then you can do a sanity check in the wrapper level, and let the adapter implementors to simply use the prop.

lherman-cs, 2020-02-18:
As you have suggested in #69, it would be nice to have an interface from the wrapper to get the default value. 

But, I think instead of putting the interface in Wrapper, we can put it under `prop` since the default values shouldn't be dependent on anything.

So, we probably have something like,

```go
func (p Media) Default() prop.Media
```
Or probably even have it in package level

```
package prop

var Default = prop.Media{
  ...
}
```

What do you think?

at-wat, 2020-02-18:
I think the default values (especially continuous parameters) depends on the actual device.
For example camera would have default framerate of 30fps in most case. But, screen capture maybe 10fps since it's too heavy at 30fps.

lherman-cs, 2020-02-18:
I see. If that's the case, I'm not seeing that we can do better than we already have rn, set the default value in each adapter implementation.

closed 2020-02-20 by lherman-cs