pion/mediadevices #119: Update codec builder to support rate limiting while running

lherman-cs, 2020-03-24:
As of now, `VideoEncoderBuilder` and `AudioEncoderBuilder` only return `io.ReadCloser` and `error:

```go
BuildAudioEncoder(r audio.Reader, p prop.Media) (io.ReadCloser, error)
BuildVideoEncoder(r video.Reader, p prop.Media) (io.ReadCloser, error)
```

While returning `io.ReadCloser` is very idiomatic, `io.ReadCloser` is not enough for our need. The main limitation is rate-limiting, we can't adjust the codec parameters on the fly or decreasing/increasing the bitrate as needed depending on the current network speed and quality.

So, instead of returning `io.ReadCloser`, it's better to instead return a new interface that embeds `io.ReadCloser` and has another method that updates the BaseParam:

```go
package codec

import "io"

type ReadCloser interface {
	io.ReadCloser
	Update(params BaseParams) error
}
```

lherman-cs, 2020-03-24:
@at-wat what do you think about the interface?

at-wat, 2020-03-24:
I think updating `BaseParams.KeyFrameInterval` is less meaningful on WebRTC (and also pure RTP) in which RTCP PictureLossIndication packet triggers a key frame.
I'm not very sure it's better to update whole `BaseParams`.

lherman-cs, 2020-03-24:
From my understanding, setting `BaseParams.KeyFrameInterval` to 0 means the same thing with force keyframe. Did I misunderstand this?

at-wat, 2020-03-24:
I think `BaseParams.KeyFrameInterval=0` means that all frames are key frame.
PictureLossIndication makes next single frame into key frame.

lherman-cs, 2020-03-24:
Ah, that makes total sense. I guess nvm then, I think we can use what you've suggested already,

```go
package codec

import "io"

type ReadCloser interface {
	io.ReadCloser
	SetBitRate(int) error
	ForceKeyFrame() error
}
```

What do you think about the naming? I changed it from `StreamReader` to `ReadCloser` because I think it's more common and it's clear that's different from `io.ReadCloser`, because it'll be called with the namespace `codec.ReadCloser`. This pattern also exists in the standard library, there are [image.Image](https://golang.org/pkg/image/#Image) and [draw.Image](https://golang.org/pkg/image/draw/#Image). The main difference is that `draw.Image` has an extra method that allows setting a single pixel to a color.

at-wat, 2020-03-24:
`codec.ReadCloser` sounds good to me!

at-wat, 2020-05-18:
seems related: https://github.com/pion/webrtc/wiki/MediaAPI#rtpsender-will-have-callbacks-for-rtcp-feedback-results

EmrysMyrddin, 2021-05-11:
Is this still in progress ?

I came across this issue will trying to understand how PLI packets are handled in pion's webrtc. From what I understand of the code, receiving a PLI should trigger a new key frame, but this is not implemented for now right ?

Related to this, I have made a PR to implement key frame request for VPX: https://github.com/pion/mediadevices/pull/334

EmrysMyrddin, 2022-07-25:
This is stale and no longer applicable since #338 has been merged.

closed 2022-07-25 by EmrysMyrddin