pion/mediadevices #142: Reduce mediadevices complexity

lherman-cs, 2020-04-19:
As of now, `mediadevices` uses many interfaces:
* [MediaDevices](https://pkg.go.dev/github.com/pion/mediadevices?tab=doc#MediaDevices)
* [MediaStream](https://pkg.go.dev/github.com/pion/mediadevices?tab=doc#MediaStream)
* [Tracker](https://pkg.go.dev/github.com/pion/mediadevices?tab=doc#Tracker)
* [LocalTrack](https://pkg.go.dev/github.com/pion/mediadevices?tab=doc#LocalTrack)

While using interfaces makes the design very flexible, it doesn't give that benefit for free. Following are some of the downsides:
* Create some boilerplates and reduce maintainability because every time interfaces don't have implementation details. So, we'll always end up with 1 type for the interface and another type for the struct.
* Increase API complexity if we overuse interfaces. Every time we use an interface, it adds a layer of indirection to the actual definition, which is ok if they're only a few defined interfaces. But, the problem quickly arises when we have multiple of them and they're spread all over the place in a package.
* Reduce docs readability. I think this is similar to the second point. Since using interfaces adds an extra layer of indirection, it requires more thought process to the reader.

While I've laid out some of the downsides of using interfaces above, I still think that they're great and should be used appropriately. So, I think we should try to get rid of some of the interfaces and replace them with structs, we should remove the ones that don't require flexibilities. 

In my opinion, we should convert `MediaDevices` and `MediaStream` interfaces to structs.

Note: Hopefully when [pion/webrtc v3](https://github.com/pion/webrtc/issues/9) is ready, `Tracker` and `LocalTrack` interfaces can get merged to `pion/webrtc`.

lherman-cs, 2020-04-19:
@at-wat Do you have some thoughts on this?

EmrysMyrddin, 2022-07-25:
Do you still think this is something to do ? Do you have any new though now that pion/webrtc v3 is out ?