pion/mediadevices #146: Decouple from webrtc stuff

lherman-cs, 2020-04-19:
I think we should try to decouple `mediadevices` from webrtc stuff so that it can be more generic and useful for a wider audience. Also, if we look at the original definition of the MediaDevices API from Mozilla, they never mentioned that the API is solely for webrtc:

> The MediaDevices interface provides access to connected media input devices like cameras and microphones, as well as screen sharing. In essence, it lets you obtain access to any hardware source of media data.

Reference: https://developer.mozilla.org/en-US/docs/Web/API/MediaDevices

lherman-cs, 2020-04-19:
I think instead of using RTP or webrtc concepts, we should use Blob, https://developer.mozilla.org/en-US/docs/Web/API/Blob since this is much more generic. Then, later, we can simply extend Blob to other kinds such as RTP and webrtc.

lherman-cs, 2020-04-30:
I've created a small design draft, https://github.com/pion/mediadevices/wiki/Source-Sink-Design.

@at-wat do you mind to take a look?

at-wat, 2020-05-01:
Putting `GetUserMedia` directly under `mediadevices` package sounds reasonable.

I'm not very sure who have to own video/audio encoder.
In the new design draft, `PeerConnection` seems having encoder, but I think encoding is out of `PeerConnection`'s role.
We may want to put `extwebrtc` features to `pion/webrtc/v3`, so `PeerConnection` is better to be separated from media encoding things?

lherman-cs, 2020-05-01:
I'm definitely 100% with you. The encoding part has been really awkward. I'm not really sure either where to put it...

If we put it alongside GetUserMedia, it would feel weird since GetUserMedia is supposed to get a source based on the constraint and the result is supposed to be transformable (if we already encoded, I don't know how users can transform it)

On the other hand, if we put it in ext/webrtc, users can easily transform the source from GetUserMedia and GetUserMedia feels less awkward, and we have all the information necessary to pick the right codec from PeerConnection. But, like you said, this will make it really hard to merge it to main webrtc package, and it's probably not PeerConnection's responsibility.

So, I think the main problem is where to put the logic to decide which codec should be used. 

Maybe, we have a middle approach? Not in GetUserMedia and not in PeerConnection, but a new API?

lherman-cs, 2020-05-01:
After reading some of Chromium codebase, they put the encoder factories in their mediaengine, https://chromium.googlesource.com/external/webrtc/+/refs/heads/master/media/engine/webrtc_media_engine.h, which I think it should be similar to [Pion's mediaengine](https://pkg.go.dev/github.com/pion/webrtc?tab=doc#MediaEngine).

This definitely makes sense! But, the problem is that we can't really do that... Since `mediadevices` has a lot of cgo, which will make the main webrtc package to be not pure Go anymore.

1 possibility I can think of is that we probably should extend our media engine to also accept an encoder factory interface?

at-wat, 2020-05-01:
Sounds good.

So, if we have them in `MediaEngine`, [RTPCodec](https://pkg.go.dev/github.com/pion/webrtc/v2@v2.2.7?tab=doc#RTPCodec) may have interfaces of video/audio encoder/decoder and `Track.ReadBlob/WriteBlob` (`Blob` may be better to be replaced by `Frame` or something?) can be used if encoder/decoder is specified.

lherman-cs, 2020-05-02:
I'm not sure which RTPCodec interface that you were referring, did you mean the interfaces from V3, https://github.com/pion/webrtc/wiki/PlanningV3?

Yea, I thought having Blob concept too, at the end it became a simple writer/reader but with some meta data. Not sure, what kind of interfaces should we pass to the encoder factory/builder. The thing is the main webrtc package needs to understand frames (for video) and chunks (for audio) so that it can use the encoder to encode the frames and chunks from tracks.

Now, the question is, should the webrtc package depends on mediadevices package? Or the other way around? Or one of them defines its own interface and the other package will implement the interfaces implicitly? (Similar to packages that implements io.Reader and io.Writer implicitly)

The problem with the last approach is that the interface should only use primitive data types, otherwise there will be a dependency on the data structure itself (this itself would forbid us to not have a direct dependency).

lherman-cs, 2020-05-02:
@at-wat just updated the draft again. PTAL ^^

at-wat, 2020-05-03:
The draft looks nice.
Maybe `mediadevices.Track` will have `ReadBlob()` and wrapped by encoder to provide `ReadRTP()`?
```go
type Track interface {
  ID() string
  Stop()
  OnEnded(func(error))
  ReadBlob() ([]byte, error) // or
  ReadBlob([]byte) (int, error)
}
```

lherman-cs, 2020-05-03:
That's definitely possible. But, I'm not sure about the use case of having `ReadBlob` in `Track`. In my head, people would probably use the decoded version which they can get by casting `Track` to either `VideoTrack` or `AudioTrack`, or they would get the encoded version which they woud get from the encoder.

lherman-cs, 2020-05-06:
@at-wat I just created a POC here, https://github.com/pion/mediadevices/blob/refractor/examples/simple/main.go. Please feel free to drop feedback!

The example basically reads frames from the camera, generates MJPEGs, and serves them through HTTP. The port is `1313`.

I think it really shows:
1. It's a lot easier to use
2. It can be used for non-WebRTC
3. It's pretty easy to broadcast

lherman-cs, 2020-05-14:
It seems that there's a lot of work that needs to be done and remove a lot of things. So, I'll create a new branch called `redesign`.

at-wat, 2020-05-14:
Sorry, I just saw https://github.com/pion/mediadevices/blob/refractor/examples/simple/main.go.
Seems very nice!

lherman-cs, 2020-05-15:
Glad you like it too!

lherman-cs, 2020-05-15:
I have another commit to add the webrtc extension, https://github.com/pion/mediadevices/pull/164/commits/21fd8c984641e7fe9b9564ea35f0bd47c8cf3435. I also add a new webrtc example with the new design.

It definitely still has some rough edges, especially about the cleanup for the rtp reader. Please let me know what you think. Also, feel free to submit a PR to propose a change to that branch. I'll be very happy to look.

lherman-cs, 2020-07-01:
@at-wat, I've updated the design doc. Now, the design covers the common use case for Pion where tracks always give encoded version of media data. While covering this, I also made sure that the encoding part is not a part of track.

So, for WebRTC use cases, it'll look like the following:

```
package main

import (
       "github.com/pion/webrtc/v2"
       "github.com/pion/mediadevices"
       "github.com/pion/mediadevices/pkg/codec/x264"
)

func main() {
       ...
       s, err := mediadevices.GetUserMedia(mediadevices.MediaStreamConstraints{
		Audio: func(c *mediadevices.MediaTrackConstraints) {},
		Video: func(c *mediadevices.MediaTrackConstraints) {
			c.Width = 640
			c.Height = 480
		},
	})

        x264Params, _ := x264.NewParams()
        enhancedPC := EnhancePeerConnection(pc, &x264Params)

        for _, tracker := range s.GetTracks() {
		_, err = enhancedPC.EnhancedAddTransceiverFromTrack(tracker),
 			webrtc.RtpTransceiverInit{
				Direction: webrtc.RTPTransceiverDirectionSendonly,
			},
		)
	}
}
```

As you can see here, we can add our methods to PeerConnection by prefixing them with "Enhanced". This way, we can bring our own data structure while still providing clean APIs. What do you think?

at-wat, 2020-07-05:
https://github.com/pion/mediadevices/wiki/Source-Sink-Design#design
> // step 1: find common supported codecs from pc.opt and mediaengine
> // note 1.1: if there are multiple codecs as the result, try to build in sequential order, if one fail, use the next ones.
 
> // step 2: create a local track using the encoder builder

> // step 2: create a new RTPSender

I'm not very sure that we can create a new RTPSender outside webrtc package. RTPSender requires SRTP master key which is from DTLS session.

> // step 3: replace the RTPSender's local track from step 3 with the local track from step 2

lherman-cs, 2020-07-05:
I think it's possible. I'm thinking to just call `AddTransceiverFromTrack` which should give us an indirect access to the dtls transport. (When I said to create RTPSender, it's not through NewRTPSender but through AddTransceiverFromTrack)

When v3 is ready, the track will become an interface. But, this is okay, since we can provide a proxy implementation.

I think the idea here is only to hide the codec matching and encoding, but the rest will still use existing APIs since they can handle encoded data well already.

at-wat, 2020-07-06:
I feel that if it can be implemented as a track wrapper, the interface would be bit elegant.
Does it require to wrap PeerConnection?
```go

func main() {
       ...
       s, err := mediadevices.GetUserMedia(mediadevices.MediaStreamConstraints{
		Audio: func(c *mediadevices.MediaTrackConstraints) {},
		Video: func(c *mediadevices.MediaTrackConstraints) {
			c.Width = 640
			c.Height = 480
		},
	})

        x264Params, _ := x264.NewParams()
        rtpTracker, _ := mediadevices.NewRTPTracker(x264Params, ...(more codecs))

        for _, tracker := range s.GetTracks() {
		_, err = pc.AddTransceiverFromTrack(rtpTracker.Track(tracker)),
 			webrtc.RtpTransceiverInit{
				Direction: webrtc.RTPTransceiverDirectionSendonly,
			},
		)
	}
}
```

lherman-cs, 2020-07-06:
I'm not sure if this is possible. Currently, it's not clear if the new track interface will have access to the negotiated codec. Also, we need to register our codecs too so that PeerConnection can negotiate the available codecs with the other peer.

at-wat, 2020-07-06:
Umm, I see. It seems not only a problem of mediadevices.
https://github.com/pion/webrtc/blob/master/examples/broadcast/main.go
For example, broadcast example assumes that subscribers always support codec used by the broadcaster, but this also should have a codec negotiation logic.  (in this case, it should just validate that broadcasted codec is supported or not.)
And, payload type (number) determination is also a part of codec negotiation, which was done by application code via `MediaEngine`.

lherman-cs, 2020-07-06:
Right, so that's why I'm thinking to wrap PeerConnection so that we can register our codecs. AFAIK, we're trying to remove MediaEngine in v3. So, I assume that it'll somehow get exposed in PeerConnection? 

I don't think the new track interface will expose the codec registration since it needs to be done at the initiation time.

at-wat, 2020-07-06:
In WebAPI, it seems done in the layer of RTCRtpReceiver and RTCRtpSender.

https://developer.mozilla.org/en-US/docs/Web/API/RTCRtpReceiver#Methods
> ### `RTCRtpReceiver.getParameters()`
>    Returns an `RTCRtpParameters` object which contains information about how the RTC data is to be decoded.

https://developer.mozilla.org/en-US/docs/Web/API/RTCRtpSender#methods
> ### `RTCRtpSender.getParameters()`
>    Returns a `RTCRtpParameters` object describing the current configuration for the encoding and transmission of media on the track.
> ### `RTCRtpSender.setParameters()`
>    Applies changes to parameters which configure how the track is encoded and transmitted to the remote peer.

Parameters given to `setParameters` contain encoding information.
https://developer.mozilla.org/en-US/docs/Web/API/RTCRtpSendParameters/encodings

lherman-cs, 2020-07-07:
Wow! I didn't know this existed. In this, we can put the negotiation part in webrtc! Borrowing your idea, the flow should look like,

```
func main() {
       ...
       s, err := mediadevices.GetUserMedia(mediadevices.MediaStreamConstraints{
		Audio: func(c *mediadevices.MediaTrackConstraints) {},
		Video: func(c *mediadevices.MediaTrackConstraints) {
			c.Width = 640
			c.Height = 480
		},
	})

        x264Params, _ := x264.NewParams()
        rtpTracker, _ := mediadevices.NewRTPTracker(x264Params, ...(more codecs))

        for _, tracker := range s.GetTracks() {
                // peerconnection should also add codec availabilities under the hood for this particular track by using RTCRtpSender.getParameters()
		_, err = pc.AddTransceiverFromTrack(rtpTracker.Track(tracker)),
 			webrtc.RtpTransceiverInit{
				Direction: webrtc.RTPTransceiverDirectionSendonly,
			},
		)
	}

        // peerconnection should negotiate with the other peer
        // peerconnection should set the negotiated codec with RTCRtpSender.setParameters()
}
```

at-wat, 2020-07-25:
Umm, in my understanding, RTPSender is created by PeerConnection in WebAPI. In this case, we need to customize RTPSender to receive negotiated codec informations. It can be specified via RTPTransceiverInit as a pion extension.

at-wat, 2020-07-25:
https://github.com/pion/webrtc-v3-design/issues/10

lherman-cs, 2020-11-06:
Implemented in https://github.com/pion/mediadevices/commit/2f5e4ee9148194d1be6519e88444ba518011d5b6

closed 2020-11-06 by lherman-cs