pion/mediadevices #195: Native Client using gio

joe-getcouragenow, 2020-07-01:
Its possible to build native desktop and mobile clients using gio and i think its a great opportunity.

It can also build wasm web clients. Tinygo is close but not yet. 
Run demo here: https://gioui.org/
- type in editor
- type where it says hint
- click transform checkbox.

The work for multimedia is going on here:
https://todo.sr.ht/~eliasnaur/gio/125
- patches for video going in now.

Audio works now on desktop.
https://github.com/dejadejade/giox
cd /examples/player && go build -o player .
./player file_example_MP3_5MG.mp3

Canvas also works
https://github.com/ajstarks/giocanvas
cd play && go build -o play
./play

lherman-cs, 2020-07-01:
@joe-getcouragenow Hey, this is awesome!! Mediadevices is currently WebRTC oriented. However, we're working on supporting non-webrtc cases, https://github.com/pion/mediadevices/blob/redesign/examples/simple/main.go. 

I think the new API will be really clean!! And will be super powerful to be combined with GIO. I can imagine that people will be able to build a video/audio player and even video conferencing apps easily with this combination.

Just to give you a glimpse, I can imagine that it would be something like,

```go
	s, err := mediadevices.GetUserMedia(mediadevices.MediaStreamConstraints{
		Video: func(p *prop.Media) {
			p.Width = 600
			p.Height = 400
		},
	})
	if err != nil {
		panic(err)
	}

	t := s.GetVideoTracks()[0]
	defer t.Stop()
	videoTrack := t.(*mediadevices.VideoTrack)

        videoReader := videoTrack.NewReader()
        for {
              frame, _ := videoReader.Read()
              // render this with gio
        }
```

joe-getcouragenow, 2020-07-01:
@lherman-cs thank you so much for responding to this. 

Well we need this for WebRTC conferencing and just playing videos. 
We currently have no way to run ion webrtc outside a browser as Apple Webview blocks access to the web cam and mic on Desktop.
See here: https://github.com/webview/webview/issues/368

So desperation being the mother of all invention, i looked at Gio again, and it has really grown up.

If you like i can try to get the your WebRTC and the Gio Multimedia code together as a PR.

lherman-cs, 2020-07-01:
@joe-getcouragenow yes please, that would be awesome! Are you planning to make it as a library/example?

joe-getcouragenow, 2020-07-11:
> @joe-getcouragenow yes please, that would be awesome! Are you planning to make it as a library/example?

Both. 

The Texturebinding and Shader code patches have not been merged yet into main tip. Looks like they will be very soon, and then it should be possible to have a crack at it.

Latest Patches: https://lists.sr.ht/~eliasnaur/gio-patches/%3C159407480987.28969.4368210228585413349-0%40git.sr.ht%3E

Main tip: https://git.sr.ht/~eliasnaur/gio/tree

gedw99, 2020-12-28:
https://github.com/gioui/gio

https://twitter.com/giouiorg/status/1338855232508362752

they now accept pr's over github so its much easier to work on now.

gedw99, 2021-06-23:
looks like someone attempted video and gio here: https://github.com/dejadejade/giox/blob/master/player/video.go

I am on mac , not windows, but will see if i can get any progress....