pion/mediadevices #252: Use of Git LFS in repo causes it to be not go-get'able when you bust your monthly Github LFS bandwidth quota

darkvertex, 2020-11-17:
This is what happens when too many people git-clone'd this public repo:
```
go get github.com/pion/mediadevices
# cd .; git clone -- https://github.com/pion/mediadevices /home/alan/go/src/github.com/pion/mediadevices
Cloning into '/home/alan/go/src/github.com/pion/mediadevices'...
Downloading img/demo.gif (15 MB)
Error downloading object: img/demo.gif (bdf9bb3): Smudge error: Error downloading img/demo.gif (bdf9bb3a8d4cbcbc36f9dc4cbd1517e9b9db90081525f500ff095318fc0954b1): batch response: This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.

Errors logged to /home/alan/go/src/github.com/pion/mediadevices/.git/lfs/objects/logs/20201117T121619.060488741.log
Use `git lfs logs last` to view the log.
error: external filter 'git-lfs filter-process' failed
fatal: img/demo.gif: smudge filter lfs failed
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry with 'git restore --source=HEAD :/'

package github.com/pion/mediadevices: exit status 128
```
`go get` fails, but not because the code is bad but because Git LFS is used for some images and the quota is maxed, so should be easy to fix.

lherman-cs, 2020-11-17:
@darkvertex Ouch, I didn't know this. Thanks for reporting it! I'm going to fix it ASAP.

lherman-cs, 2020-11-17:
@darkvertex I've removed the big image. It should be working properly now.

darkvertex, 2020-11-17:
Seems good now! :+1:

Thank you for the quick fix.

closed 2020-11-17 by darkvertex