pion/mediadevices #260: Screen capture on Linux

celestial2, 2020-12-07:
I'm trying to start screen capture on Linux ( Dockerized 18.04 version ), but i keep getting the "Failed to find the best driver that fits the constraints". I've installed all of the packages and removed the constraints so that it has any on all of them. But still it fails. 

I tried it also on linux that has desktop manager installed, but no luck. I have the same app in nodejs and it catches the display successfully with the mediadevices there.
Has anyone done this setup before or have any tip of what i am missing ?

lherman-cs, 2020-12-07:
@celestial2 have you tried the following steps. https://github.com/pion/mediadevices#failed-to-find-the-best-driver-that-fits-the-constraints? What's the output?

Make sure that you imported `import _ github.com/pion/mediadevices/pkg/driver/screen`, and uses `GetDisplayMedia` instead of `GetUserMedia`. 

Out of curiosity, which library are you using to enable mediadevices in nodejs?

celestial2, 2020-12-07:
You saved me with the GetDisplayMedia, it works like charm now.
I am using the one from the navigator: `navigator.mediaDevices.getUserMedia()`

closed 2020-12-07 by celestial2