pion/mediadevices #270: error: '_vsnprintf' was not declared in this scope

rasen0, 2020-12-16:
### Your environment.
  - Version: *Windows 10home  20H2*

### code
```golang

import (
	"image/jpeg"
	"os"

	"github.com/pion/mediadevices"
	"github.com/pion/mediadevices/pkg/prop"

	// This is required to register camera adapter
	_ "github.com/pion/mediadevices/pkg/driver/camera"
	// Note: If you don't have a camera or your adapters are not supported,
	//       you can always swap your adapters with our dummy adapters below.
	// _ "github.com/pion/mediadevices/pkg/driver/videotest"
)

func main() {
	stream, _ := mediadevices.GetUserMedia(mediadevices.MediaStreamConstraints{
		Video: func(constraint *mediadevices.MediaTrackConstraints) {
			// Query for ideal resolutions
			constraint.Width = prop.Int(600)
			constraint.Height = prop.Int(400)
		},
	})

	// Since track can represent audio as well, we need to cast it to
	// *mediadevices.VideoTrack to get video specific functionalities
	track := stream.GetVideoTracks()[0]
	videoTrack := track.(*mediadevices.VideoTrack)
	defer videoTrack.Close()

	// Create a new video reader to get the decoded frames. Release is used
	// to return the buffer to hold frame back to the source so that the buffer
	// can be reused for the next frames.
	videoReader := videoTrack.NewReader(false)
	frame, release, _ := videoReader.Read()
	defer release()

	// Since frame is the standard image.Image, it's compatible with Go standard
	// library. For example, capturing the first frame and store it as a jpeg image.
	output, _ := os.Create("frame.jpg")
	jpeg.Encode(output, frame, nil)
}

```

### console
```
# github.com/pion/mediadevices/pkg/driver/camera
In file included from /usr/include/w32api/dshow.h:33,
                 from camera_windows.cpp:4:
/usr/include/w32api/strsafe.h: In function 'HRESULT StringVPrintfWorkerA(STRSAFE_LPSTR, size_t, STRSAFE_LPCSTR, va_list)':
/usr/include/w32api/strsafe.h:1555:12: error: '_vsnprintf' was not declared in this scope
 1555 |     iRet = _vsnprintf(pszDest,cchMax,pszFormat,argList);
      |            ^~~~~~~~~~
/usr/include/w32api/strsafe.h:1555:12: note: the macro '_vsnprintf' had not yet been defined
/usr/include/w32api/strsafe.h:1961: note: it was later defined here
 1961 | #define _vsnprintf _vsnprintf_instead_use_StringCbVPrintfA_or_StringCchVPrintfA;
      | 
/usr/include/w32api/strsafe.h: In function 'HRESULT StringVPrintfWorkerW(STRSAFE_LPWSTR, size_t, STRSAFE_LPCWSTR, va_list)':
/usr/include/w32api/strsafe.h:1575:12: error: '_vsnwprintf' was not declared in this scope
 1575 |     iRet = _vsnwprintf(pszDest,cchMax,pszFormat,argList);
      |            ^~~~~~~~~~~
/usr/include/w32api/strsafe.h:1575:12: note: the macro '_vsnwprintf' had not yet been defined
/usr/include/w32api/strsafe.h:1964: note: it was later defined here
 1964 | #define _vsnwprintf _vsnwprintf_instead_use_StringCbVPrintfW_or_StringCchVPrintfW;
      | 
/usr/include/w32api/strsafe.h: In function 'HRESULT StringVPrintfExWorkerA(STRSAFE_LPSTR, size_t, size_t, char**, size_t*, unsigned int, STRSAFE_LPCSTR, va_list)':
/usr/include/w32api/strsafe.h:1612:9: error: '_vsnprintf' was not declared in this scope
 1612 |  iRet = _vsnprintf(pszDest,cchMax,pszFormat,argList);
      |         ^~~~~~~~~~
/usr/include/w32api/strsafe.h:1612:9: note: the macro '_vsnprintf' had not yet been defined
/usr/include/w32api/strsafe.h:1961: note: it was later defined here
 1961 | #define _vsnprintf _vsnprintf_instead_use_StringCbVPrintfA_or_StringCchVPrintfA;
      | 
/usr/include/w32api/strsafe.h: In function 'HRESULT StringVPrintfExWorkerW(STRSAFE_LPWSTR, size_t, size_t, wchar_t**, size_t*, unsigned int, STRSAFE_LPCWSTR, va_list)':
/usr/include/w32api/strsafe.h:1685:9: error: '_vsnwprintf' was not declared in this scope
 1685 |  iRet = _vsnwprintf(pszDest,cchMax,pszFormat,argList);
      |         ^~~~~~~~~~~
/usr/include/w32api/strsafe.h:1685:9: note: the macro '_vsnwprintf' had not yet been defined
/usr/include/w32api/strsafe.h:1964: note: it was later defined here
 1964 | #define _vsnwprintf _vsnwprintf_instead_use_StringCbVPrintfW_or_StringCchVPrintfW;
      | 
```

lherman-cs, 2020-12-16:
@rasen0 what environment are you using? Is it Cygwin or Mingw? If Cygwin or other environments, could you try to use Mingw-x64?

I've tested using Mingw-x64 on my VM.

lherman-cs, 2021-02-21:
Due to inactivity, I'm going to close this issue. Please feel free to reopen it.

closed 2021-02-21 by lherman-cs

PaiGack, 2022-08-13:
linux 上无该函数，
error: ‘_vscprintf’ was not declared in this scope; did you mean ‘vswprint