pion/mediadevices #287: Camera capture doesn't work on MacOS

andrein, 2021-01-12:
### Your environment.
  - Version: 0.1.14, also tried master.
  - Operating system: MacOS 11.1

### What did you do?
Ran the webrtc and facedetection examples.

### What happened?
Both failed with the same panic:

```
❯ ./webrtc < sdp

mediadevices DEBUG: 00:18:32.060105 mediadevices.go:151:
============ Found Properties ============

=============== Constraints ==============

DeviceID: any
VideoConstraints:
  Width: 640 (ideal)
  Height: 480 (ideal)
  FrameRate: any
  FrameFormat: I420 (ideal)
AudioConstraints:
  ChannelCount: any
  Latency: any
  SampleRate: any
  SampleSize: any
  IsBigEndian: any
  IsFloat: any
  IsInterleaved: any

================ Best Fit ================

Not found
panic: failed to find the best driver that fits the constraints

goroutine 1 [running]:
main.main()
	/Users/andrein/Projects/pion/mediadevices/examples/webrtc/main.go:83 +0x8b3
```

```
❯ ./facedetection
mediadevices DEBUG: 00:11:08.300393 mediadevices.go:151:
============ Found Properties ============

=============== Constraints ==============

DeviceID: any
VideoConstraints:
  Width: 640 (ideal)
  Height: 480 (ideal)
  FrameRate: any
  FrameFormat: I420 (exact)
AudioConstraints:
  ChannelCount: any
  Latency: any
  SampleRate: any
  SampleSize: any
  IsBigEndian: any
  IsFloat: any
  IsInterleaved: any

================ Best Fit ================

Not found
panic: failed to find the best driver that fits the constraints
```

@TannerGabriel suggested that the constraints might be too strict, so I removed all of them (width, height, frame format).
I've also tried setting every frame format from https://github.com/pion/mediadevices/blob/master/pkg/frame/const.go without success.

Tried debugging the examples and noticed that Session.Properties() returns null. 

I'm wondering if this might be a MacOS Security & Privacy issue. I didn't get the "allow access to the camera" prompt when running the examples.

lherman-cs, 2021-01-13:
@andrein thanks for reporting. Unfortunately, I don't own a Mac laptop anymore, I can't test this. My best guess would be that your camera's format is not included in this mapping, https://github.com/pion/mediadevices/blob/master/pkg/avfoundation/AVFoundationBind/AVFoundationBind.m#L134-L164. 

It seems like you can use wacaw for v4l2-ctl equivalent on mac, https://stackoverflow.com/questions/19038706/video4linux2-command-for-mac-osx-with-isight. If this doesn't work, probably add a classic print statement ^^. Since this is objective C, you can use either printf or NSLog.

andrein, 2021-01-13:
@lherman-cs I've tried wacaw, but it looks like it's only built for i386 and ppc so I couldn't launch it since a few versions ago MacOS removed support for 32 bit software IIRC. 

I've scoured the interwebs and found you can get the pixel formats using ffmpeg:
```
❯ ffmpeg -f avfoundation -framerate 30 -i default
ffmpeg version 4.3.1 Copyright (c) 2000-2020 the FFmpeg developers
  built with Apple clang version 12.0.0 (clang-1200.0.32.28)
  configuration: --prefix=/usr/local/Cellar/ffmpeg/4.3.1_8 --enable-shared --enable-pthreads --enable-version3 --enable-avresample --cc=clang --host-cflags= --host-ldflags= --enable-ffplay --enable-gnutls --enable-gpl --enable-libaom --enable-libbluray --enable-libdav1d --enable-libmp3lame --enable-libopus --enable-librav1e --enable-librubberband --enable-libsnappy --enable-libsrt --enable-libtesseract --enable-libtheora --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libxvid --enable-lzma --enable-libfontconfig --enable-libfreetype --enable-frei0r --enable-libass --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-librtmp --enable-libspeex --enable-libsoxr --enable-videotoolbox --enable-libzmq --enable-libzimg --disable-libjack --disable-indev=jack
  libavutil      56. 51.100 / 56. 51.100
  libavcodec     58. 91.100 / 58. 91.100
  libavformat    58. 45.100 / 58. 45.100
  libavdevice    58. 10.100 / 58. 10.100
  libavfilter     7. 85.100 /  7. 85.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  7.100 /  5.  7.100
  libswresample   3.  7.100 /  3.  7.100
  libpostproc    55.  7.100 / 55.  7.100
[avfoundation @ 0x7f8e3a808200] Selected pixel format (yuv420p) is not supported by the input device.
[avfoundation @ 0x7f8e3a808200] Supported pixel formats:
[avfoundation @ 0x7f8e3a808200]   uyvy422
[avfoundation @ 0x7f8e3a808200]   yuyv422
[avfoundation @ 0x7f8e3a808200]   nv12
[avfoundation @ 0x7f8e3a808200]   0rgb
[avfoundation @ 0x7f8e3a808200]   bgr0
[avfoundation @ 0x7f8e3a808200] Overriding selected pixel format to use uyvy422 instead.
Input #0, avfoundation, from 'default':
  Duration: N/A, start: 21857.456233, bitrate: N/A
    Stream #0:0: Video: rawvideo (UYVY / 0x59565955), uyvy422, 1280x720, 30 tbr, 1000k tbn, 1000k tbc
At least one output file must be specified
```

How can I map the pixel formats reported by ffmpeg to their avfoundation counterparts?

PR #289 (2021-01-15): Add AVBindFrameFormatYUY2 mapping in AVFoundationBind
#### Description
Attempt to fix the MacOS integrated camera pixel format issues.

The integrated camera on my 2018 Macbook Pro supports the following pixel formats:
```
[avfoundation @ 0x7f8e3a808200] Supported pixel formats:
[avfoundation @ 0x7f8e3a808200]   uyvy422
[avfoundation @ 0x7f8e3a808200]   yuyv422
[avfoundation @ 0x7f8e3a808200]   nv12
[avfoundation @ 0x7f8e3a808200]   0rgb
[avfoundation @ 0x7f8e3a808200]   bgr0
```

I was only able to find the correct mapping for the YUY2 pixel format and it fixed my issue. If anyone knows the o [... cut]
closed 2021-01-15 by lherman-cs