pion/mediadevices #290: Don't work camera in mac

MunDeuksoo, 2021-01-20:
### Your environment.
  - Version: Releases
  - Browser: *include version*
  - Other Information - *stacktraces, related issues, suggestions how to fix, links for us to have context*

### What did you do?

### What did you expect?
i was resulted error.

### What happened?
Error Message
:github.com/pion/mediadevices/pkg/avfoundation/avfoundation_darwin.go:6:11: fatal error: 'AVFoundationBind/AVFoundationBind.h' file not found

I use mac book pro.

I know my mac book media infomation in below.

  uyvy422
  yuyv422
  nv12
   0rgb
  bgr0

PS. Please help me.

MunDeuksoo, 2021-01-20:
I solved this problem.
When "import" to "Golang", the avfoundation bind folder itself does not enter the avfoundation folder.
When you import to Golang, tell me how to receive the folder as well.

I just added it myself.

You don't have to worry. I didn't add it to the git.

MunDeuksoo, 2021-01-20:
I added the AVFoundationBind folder because it was confirmed that there was no AVFoundationBind folder when I entered the uri below. Can you check it out and reply to me again?
[image]

lherman-cs, 2021-01-24:
@MunDeuksoo I'm not sure how this could happen as AVFoundationBind.h file is included in this repository, https://github.com/pion/mediadevices/tree/master/pkg/avfoundation/AVFoundationBind.

Would you mind sharing the reproducible steps?

donatj, 2021-02-03:
I'm getting the following while trying to build the README example. I suspect it's related?

<details>
  <summary>Error Message</summary>
```
# github.com/pion/mediadevices/pkg/avfoundation
In file included from ../../../../pkg/mod/github.com/pion/mediadevices@v0.1.14/pkg/avfoundation/avfoundation_darwin.go:7:
./AVFoundationBind/AVFoundationBind.m:179:9: error: use of undeclared identifier 'AVCaptureDeviceTypeExternalUnknown'
        AVCaptureDeviceTypeExternalUnknown
        ^
./AVFoundationBind/AVFoundationBind.m:177:9: error: 'AVCaptureDeviceTypeBuiltInWideAngleCamera' is unavailable: not available on macOS
        AVCaptureDeviceTypeBuiltInWideAngleCamera,
        ^
/System/Library/Frameworks/AVFoundation.framework/Headers/AVCaptureDevice.h:441:38: note: 'AVCaptureDeviceTypeBuiltInWideAngleCamera' has been explicitly marked unavailable here
AVF_EXPORT AVCaptureDeviceType const AVCaptureDeviceTypeBuiltInWideAngleCamera API_AVAILABLE(ios(10.0)) API_UNAVAILABLE(macos) __WATCHOS_PROHIBITED __TVOS_PROHIBITED;
                                     ^
In file included from ../../../../pkg/mod/github.com/pion/mediadevices@v0.1.14/pkg/avfoundation/avfoundation_darwin.go:7:
./AVFoundationBind/AVFoundationBind.m:178:9: error: 'AVCaptureDeviceTypeBuiltInMicrophone' is unavailable: not available on macOS
        AVCaptureDeviceTypeBuiltInMicrophone,
        ^
/System/Library/Frameworks/AVFoundation.framework/Headers/AVCaptureDevice.h:435:38: note: 'AVCaptureDeviceTypeBuiltInMicrophone' has been explicitly marked unavailable here
AVF_EXPORT AVCaptureDeviceType const AVCaptureDeviceTypeBuiltInMicrophone API_AVAILABLE(ios(10.0)) API_UNAVAILABLE(macos) __WATCHOS_PROHIBITED __TVOS_PROHIBITED;
                                     ^
In file included from ../../../../pkg/mod/github.com/pion/mediadevices@v0.1.14/pkg/avfoundation/avfoundation_darwin.go:7:
./AVFoundationBind/AVFoundationBind.m:181:5: error: 'AVCaptureDeviceDiscoverySession' is unavailable: not available on macOS
    AVCaptureDeviceDiscoverySession *refSession = [AVCaptureDeviceDiscoverySession
    ^
/System/Library/Frameworks/AVFoundation.framework/Headers/AVCaptureDevice.h:1711:12: note: 'AVCaptureDeviceDiscoverySession' has been explicitly marked unavailable here
@interface AVCaptureDeviceDiscoverySession : NSObject
           ^
In file included from ../../../../pkg/mod/github.com/pion/mediadevices@v0.1.14/pkg/avfoundation/avfoundation_darwin.go:7:
./AVFoundationBind/AVFoundationBind.m:182:9: error: 'AVCaptureDeviceDiscoverySession' is unavailable: not available on macOS
        discoverySessionWithDeviceTypes: refAllTypes
        ^
/System/Library/Frameworks/AVFoundation.framework/Headers/AVCaptureDevice.h:1711:12: note: 'AVCaptureDeviceDiscoverySession' has been explicitly marked unavailable here
@interface AVCaptureDeviceDiscoverySession : NSObject
           ^
5 errors generated.
```
</details>

lherman-cs, 2021-02-09:
@donatj the error message seems to be related to a breaking change in XCode. Do you mind sharing your XCode version?

adwpc, 2021-03-13:
@donatj @lherman-cs 
https://developer.apple.com/documentation/avfoundation/AVCaptureDeviceTypeExternalUnknown
i found these errors
some variable only work on 10.15
[image]

MunDeuksoo, 2021-03-15:
> @MunDeuksoo I'm not sure how this could happen as AVFoundationBind.h file is included in this repository, https://github.com/pion/mediadevices/tree/master/pkg/avfoundation/AVFoundationBind.
> 
> Would you mind sharing the reproducible steps?

In my memory, I saw the error message and copied the part directly from avfoundtaion and added the file after removing it.

MunDeuksoo, 2021-03-15:
@lherman-cs 
In my memory, I saw the error message and copied the part directly from avfoundtaion and added the file after removing it.

http600, 2021-06-29:
> @donatj @lherman-cs
> https://developer.apple.com/documentation/avfoundation/AVCaptureDeviceTypeExternalUnknown
> i found these errors
> some variable only work on 10.15
> [image]
> we can add the `version need` in the readme

so how do we fix it other than system upgrading?

paralin, 2024-01-15:
Works when /not/ vendored, but when `go mod vendor` I get this error. See #342 as well