pion/mediadevices #292: GetMediaDevice does not work on new Macbook Pro

rivercheng, 2021-01-22:
It worked on my old Macbook Pro, but failed on my new one. 
================ Best Fit ================

Not found
panic: failed to find the best driver that fits the constraints

rivercheng, 2021-01-22:
Seems to be the camera, because if I replace the 'camera' with 'videotest', then there's no error anymore. Maybe the new Macbook pro needs a new video driver?

rivercheng, 2021-01-22:
The new Macbook pro is MacBook Pro (16-inch, 2019)

lherman-cs, 2021-01-24:
@rivercheng thanks for reporting! This issue seems to be similar to https://github.com/pion/mediadevices/issues/287. Which version of `mediadevices` are you using? Could you try to pull the latest change which contains the fix, https://github.com/pion/mediadevices/issues/287? 

If that still doesn't fix it, could you try to run `ffmpeg -f avfoundation -framerate 30 -i default`, it should give us a list of supported formats by your camera. Very likely, we just need to add another mapping from AVFoundation to mediadevices.

nrwinner, 2021-01-26:
I can confirm I am also experiencing this issue with the same computer model. I'm using the most recent version of `mediadevices`. When I run that command, I see
```
[avfoundation @ 0x7f8051814c00] Selected pixel format (yuv420p) is not supported by the input device.
[avfoundation @ 0x7f8051814c00] Supported pixel formats:
[avfoundation @ 0x7f8051814c00]   uyvy422
[avfoundation @ 0x7f8051814c00]   yuyv422
[avfoundation @ 0x7f8051814c00]   nv12
[avfoundation @ 0x7f8051814c00]   0rgb
[avfoundation @ 0x7f8051814c00]   bgr0
```

daniel-d-truong, 2021-01-27:
^ just wanted to bump as i have the same issues as well

lherman-cs, 2021-01-28:
@nrwinner thanks for the extra information! From the output, `mediadevices` support both `uyvy422` and `yuyv422`. The FFmpeg mapping can be found here, https://github.com/FFmpeg/FFmpeg/blob/c810a9502cebe32e1dd08ee3d0d17053dde44aa9/libavdevice/avfoundation.m#L53-L80. The mapping is matched to what `mediadevices` has, https://github.com/pion/mediadevices/blob/master/pkg/avfoundation/AVFoundationBind/AVFoundationBind.m#L140-L145.

In https://github.com/pion/mediadevices/pull/295, I added another nv12 support, which I hope it can detect this format in avfoundation layer. Also, I added some warnings in avfoundation so that we can see what formats are skipped. 

@daniel-d-truong @rivercheng @nrwinner could any of you try the PR? At least, if it still doesn't work, it should give you some warnings that can be useful to debug this further. Thanks!

daniel-d-truong, 2021-01-28:
@lherman-cs Just tried it and it works for me! It can read my webcam camera now. I am noticing another issue though, where when I plug in an external USB camera, the library doesn't detect that input (when I run `mediadevices.EnumerateDevices()`, only the internal laptop webcam shows. This might be better as a separate issue since this was an issue before the PR as well, but yeah just thought I'd let you know.

lherman-cs, 2021-02-03:
@daniel-d-truong Thanks for trying it! I'll keep your issue on a different thread, and will fix it in a different PR.

rivercheng, 2021-02-08:
@lherman-cs  After pulling the master, now the error is

"panic: NV12 is not supported"

rivercheng, 2021-02-08:
I think this is caused in NewDecoder() (pkg/frame/decode.go). FormatNV12 is not supported there.

nrwinner, 2021-02-08:
I'm seeing something different - after pulling the new version, `EnumerateDevices` will now show my camera(s) and `stream, err := mediadevices.GetUserMedia(mediadevices.MediaStreamConstraints{})` doesn't error out anymore, but calling `stream.GetTracks` yields an empty array 😦

nrwinner, 2021-02-08:
I'm seeing the same. I did see the warnings you added @lherman-cs, I pasted the entire trace here in the event you find it helpful. 
```
2021-02-08 09:22:16.936 test[28878:1490490] [WARNING] skipping  -2126619224x-2126619224 since it's not supported
2021-02-08 09:22:16.937 test[28878:1490490] [WARNING] skipping  -2126619224x-2126619224 since it's not supported
2021-02-08 09:22:16.937 test[28878:1490490] [WARNING] skipping  -2126619224x-2126619224 since it's not supported
2021-02-08 09:22:16.937 test[28878:1490490] [WARNING] skipping  -2126619224x-2126619224 since it's not supported
2021-02-08 09:22:16.937 test[28878:1490490] [WARNING] skipping  -2126619224x-2126619224 since it's not supported
2021-02-08 09:22:16.937 test[28878:1490490] [WARNING] skipping  -2126619224x-2126619224 since it's not supported
2021-02-08 09:22:16.937 test[28878:1490490] [WARNING] skipping  -2126619224x-2126619224 since it's not supported
2021-02-08 09:22:16.937 test[28878:1490490] [WARNING] skipping  -2126619224x-2126619224 since it's not supported
2021-02-08 09:22:16.937 test[28878:1490490] [WARNING] skipping  -2126619224x-2126619224 since it's not supported
2021-02-08 09:22:16.937 test[28878:1490490] [WARNING] skipping  -2126619224x-2126619224 since it's not supported
2021-02-08 09:22:16.937 test[28878:1490490] [WARNING] skipping  -2126619224x-2126619224 since it's not supported
2021-02-08 09:22:16.937 test[28878:1490490] [WARNING] skipping  -2126619224x-2126619224 since it's not supported
2021-02-08 09:22:16.937 test[28878:1490490] [WARNING] skipping  -2126619224x-2126619224 since it's not supported
2021-02-08 09:22:16.937 test[28878:1490490] [WARNING] skipping  -2126619224x-2126619224 since it's not supported
2021-02-08 09:22:16.937 test[28878:1490490] [WARNING] skipping  -2126619224x-2126619224 since it's not supported
2021-02-08 09:22:16.937 test[28878:1490490] [WARNING] skipping  -2126619224x-2126619224 since it's not supported
2021-02-08 09:22:16.937 test[28878:1490490] [WARNING] skipping  -2126619224x-2126619224 since it's not supported
2021-02-08 09:22:16.937 test[28878:1490490] [WARNING] skipping  -2126619224x-2126619224 since it's not supported
panic: NV12 is not supported
```

lherman-cs, 2021-02-09:
I'm sorry for the miss 😓. https://github.com/pion/mediadevices/pull/298 should fix this, also added a unit test to cover this.

lherman-cs, 2021-02-09:
@nrwinner @rivercheng do you mind trying it again?

rivercheng, 2021-02-09:
@lherman-cs  Thanks a lot. :) Now it works. However, the following error happened.

unexpected fault address 0x355e5fa0
fatal error: fault
[signal SIGSEGV: segmentation violation code=0x1 addr=0x355e5fa0 pc=0x406ac25]

goroutine 17 [running, locked to thread]:
runtime.throw(0x44b83a4, 0x5)
	/usr/local/go/src/runtime/panic.go:1116 +0x72 fp=0xc000058d70 sp=0xc000058d40 pc=0x4037d12
runtime.sigpanic()
	/usr/local/go/src/runtime/signal_unix.go:749 +0x3e5 fp=0xc000058da0 sp=0xc000058d70 pc=0x404bfc5
runtime.memmove(0xc000300000, 0x35550020, 0x96000)
	/usr/local/go/src/runtime/memmove_amd64.s:332 +0x3c5 fp=0xc000058da8 sp=0xc000058da0 pc=0x406ac25
runtime.gobytes(0x35550020, 0x96000, 0x0, 0x0, 0x0)
	/usr/local/go/src/runtime/string.go:311 +0x97 fp=0xc000058de0 sp=0xc000058da8 pc=0x4051dd7
github.com/pion/mediadevices/pkg/avfoundation._Cfunc_GoBytes(...)
	_cgo_gotypes.go:200
github.com/pion/mediadevices/pkg/avfoundation.onData.func1(0x35550020, 0x96000, 0x0, 0x40182d4, 0x7000066ec848)
	/Users/chengw/go/pkg/mod/github.com/pion/mediadevices@v0.1.15-0.20210209020436-217e634f7e24/pkg/avfoundation/avfoundation_callback_darwin.go:21 +0x59 fp=0xc000058e18 sp=0xc000058de0 pc=0x41e5bd9
github.com/pion/mediadevices/pkg/avfoundation.onData(0xc00000f6e8, 0x35550020, 0x96000)
	/Users/chengw/go/pkg/mod/github.com/pion/mediadevices@v0.1.15-0.20210209020436-217e634f7e24/pkg/avfoundation/avfoundation_callback_darwin.go:21 +0x37 fp=0xc000058e68 sp=0xc000058e18 pc=0x41e49f7
github.com/pion/mediadevices/pkg/avfoundation._cgoexpwrap_87856562e38d_onData(0xc00000f6e8, 0x35550020, 0x96000)
	_cgo_gotypes.go:220 +0x3d fp=0xc000058e90 sp=0xc000058e68 pc=0x41e499d
runtime.call32(0x0, 0x7000066ec7b0, 0x7000066ec848, 0x18)
	/usr/local/go/src/runtime/asm_amd64.s:540 +0x3e fp=0xc000058ec0 sp=0xc000058e90 pc=0x406807e
runtime.cgocallbackg1(0x0)
	/usr/local/go/src/runtime/cgocall.go:332 +0x1a5 fp=0xc000058f58 sp=0xc000058ec0 pc=0x4007425
runtime.cgocallbackg(0x0)
	/usr/local/go/src/runtime/cgocall.go:207 +0xc5 fp=0xc000058fc0 sp=0xc000058f58 pc=0x40071c5
runtime.cgocallback_gofunc(0x0, 0x0, 0x0, 0x0)
	/usr/local/go/src/runtime/asm_amd64.s:794 +0x9b fp=0xc000058fe0 sp=0xc000058fc0 pc=0x40697fb
runtime.goexit()
	/usr/local/go/src/runtime/asm_amd64.s:1374 +0x1 fp=0xc000058fe8 sp=0xc000058fe0 pc=0x4069a81

nrwinner, 2021-02-09:
It's working for me now, thanks a lot! 🤩

zyxar, 2021-02-16:
> @lherman-cs Thanks a lot. :) Now it works. However, the following error happened.
>
> unexpected fault address 0x355e5fa0
> fatal error: fault
> [signal SIGSEGV: segmentation violation code=0x1 [...]

@rivercheng 
can temporarily revert https://github.com/pion/mediadevices/commit/d86fc4a3e92f120a390079974298902465533685 
I remember the webrtc example worked when I changed my laptop to new macbook pro at the beginning of this year. Spent some time to debug, only to find the default frame format was changed to I420. And `YUY2` still works by far.

lherman-cs, 2021-02-21:
@rivercheng @nrwinner could you try the fix from https://github.com/pion/mediadevices/pull/302. Thanks to @zyxar, the issue should be fixed now.

PR #302 (2021-02-21): Fixes avfoundation panic when using NV12 frame
#### Description
This fixes panic when using `NV12` frame on macos platform.

#### Reference issue
Fixes #292
closed 2021-02-21 by lherman-cs