pion/mediadevices #312: ScalerFastBoxSampling produces incorrect results

f-fl0, 2021-03-03:
### Your environment.
  - Version: *0.1.17*

### What did you do?

```
package main

import (
	"fmt"
	"image"
	"image/jpeg"
	"os"

	"github.com/pion/mediadevices/pkg/io/video"
)

func main() {

	inFile, _ := os.Open("image.jpg")
	defer inFile.Close()

	img, _, _ := image.Decode(inFile)
	src := video.ReaderFunc(func() (image.Image, func(), error) {
		return img, func() {}, nil
	})

	scaleFBS := video.Scale(640, 480, video.ScalerFastBoxSampling)(src)
	scaled, _, _ := scaleFBS.Read()

	outFile, _ := os.Create(fmt.Sprintf("scaled.jpg"))
	defer outFile.Close()
	jpeg.Encode(outFile, scaled, nil)
}
```
where `image.jpg` is https://upload.wikimedia.org/wikipedia/commons/d/dc/DenglerSW-Parliament-Ottawa-20050730-1280x960.jpg
	
### What did you expect?
[image]

### What happened?
[image]

f-fl0, 2021-03-03:
Changing `ScalerFastBoxSampling` to `ScalerFastNearestNeighbor`, produces this image:
[image]

Using `ScalerFastBoxSampling` while removing the `if` condition from these lines https://github.com/pion/mediadevices/blob/b5b065369714d19004e7aedf18e8a186586b44a4/pkg/io/video/scaler_cgo.go#L158-L160 produces the same image.

at-wat, 2021-03-03:
What is the input [YCbCrSubsampleRatio](https://golang.org/pkg/image/#YCbCrSubsampleRatio)?

f-fl0, 2021-03-03:
> What is the input [YCbCrSubsampleRatio](https://golang.org/pkg/image/#YCbCrSubsampleRatio)?

```
	imgYCbCr := img.(*image.YCbCr)
	fmt.Println(imgYCbCr.SubsampleRatio)
	src := video.ReaderFunc(func() (image.Image, func(), error) {
		return imgYCbCr, func() {}, nil
	})
```
Print to the console: `YCbCrSubsampleRatio420` and produces the same broken outputs for both `ScalerFastBoxSampling` and `ScalerFastNearestNeighbor`.

f-fl0, 2021-03-03:
```
scaleFBS := video.Scale(640, 480, video.ScalerFastBoxSampling)(video.ToRGBA(src))
```
Leads to the desired result.

closed 2022-03-18 by at-wat