pion/mediadevices #339: PionWebRTC with Telegram have some issue

Laky-64, 2021-05-26:
### What i need?
I am trying to communicate with Telegram (Telegram Group Calls), the code works without problems, but if I try to play an audio file on Telegram I hear nothing

### The code i'm using
https://github.com/pytgcalls/gotgcalls

### The old code (I'm trying to rebuild the nodejs part in go)
https://github.com/pytgcalls/pytgcalls

#### P.S. 
The files of go part is in dist folder

### About my environment
pion/webrtc = v3.0.29
golang = 1.16.4
Platform = Windows 20H2 and Ubuntu 20.04

### What type of audio telegram need?
Need PCM16L audio

I hope for your help, and thank you in advance

Sean-Der, 2021-05-26:
Hey @at-wat @lherman-cs no rush, but curious if this is possible/how hard it would be?

I am happy to do the work myself!

StarkGang, 2021-06-12:
I totally agree, this is great idea. This Would help thousands of developers who work with telegram group call :)

lherman-cs, 2021-06-12:
Correct me if I'm wrong, you want a 16-bit raw audio format (PCM16L) and use Opus to encode.

In this case, `pion/mediadevices` supports this already. If your audio source supports the format already, you can do the following:

```go
	mediadevices.GetUserMedia(mediadevices.MediaStreamConstraints{
		Audio: func(c *mediadevices.MediaTrackConstraints) {
                        c.SampleSize = prop.IntExact(16) // <-- this will test the source to use 16 bit format.
		},
		Codec: codecSelector,
	})

```

If not, you would need to create your own converter before feeding it to the encoder, you can use https://pkg.go.dev/github.com/pion/mediadevices#AudioTrack.Transform to convert the audio format.

Laky-64, 2021-06-12:
it's possible to add the support natively? Because i need just only PCM16L Raw and not converted to Opus

lherman-cs, 2021-06-13:
@Laky-64 if you just need to read raw data (without WebRTC), you can do the following:

```go
	mediaStream, err := mediadevices.GetUserMedia(mediadevices.MediaStreamConstraints{
		Audio: func(c *mediadevices.MediaTrackConstraints) {
                        c.SampleSize = prop.IntExact(16) // <-- this will set the source to use 16 bit format.
		},
		Codec: codecSelector,
	})
        must(err)

	// since we're trying to access the raw data, we need to cast Track to its real type, *mediadevices.AudioTrak
	audioTrack := mediaStream.GetAudioTracks()[0].(*mediadevices.AudioTrack)
	defer audioTrack.Close()

	audioReader := audioTrack.NewReader(false)
     
	for  {
		chunk, release, err := audioReader.Read()
		must(err)

                 // do something with the chunk

		release()
	}

```

If you want to use it with WebRTC, you would need to implement your own AudioEncoderBuilder (https://pkg.go.dev/github.com/pion/mediadevices/pkg/codec#AudioEncoderBuilder), and register the custom codec type to pion/webrtc. This is to trick WebRTC to think that PCM16L is an encoded version (in the implementation, you just need to pass through the raw data). Since PCM16L will be a custom format, you would also need to create your RTP packetizer and depacketizer.

Laky-64, 2021-06-13:
I never done that and i don't how to do😅, i'm new in webrtc, A helping hand to build the way to use PCM16L on webrtc is appreciated, more than using the microphone (if it's that hard), I'd like to read a PCM16L file directly

rojserbest, 2021-06-18:
Greetings to you, ladies and gentlemen in @pion, including @Iherman-cs, @Sean-Der.

I would like to know if you can re-think about doing whatever called to making a connection between Golang and Telegram video/voice calls, as we, the Telegram community, aren't talented and blessed as much as your team.

We think by doing that, it will be a very big achievement for both of us, you will be the first one doing this and your names will be promoted everywhere (perhaps each of you turning into a celeb or a Hollywood star one day? who knows?)!
Plus we, will be very happy using your software to make wonderful things that will make tons of people happy.

Regards,

StarkGang, 2021-06-18:
> Greetings to you, ladies and gentlemen in @pion, including @lherman-cs and @Sean-Der.
> 
> I would like to know if you can re-think about doing whatever called to making a connection between Golang and Telegram video/voice calls, as we, the Telegram community, aren't talented and blessed as much as your team.
> 
> We think by doing that, it will be a very big achievement for both of us: you will be the first one doing this and your names will be promoted everywhere (perhaps each of you turning into a celeb or a Hollywood star one day? who knows?)!
> Plus we, will be very happy using your software to make wonderful things that will make tons of people happy.
> 
> Regards,

+1

lherman-cs, 2021-06-23:
@Laky-64 Sorry for the delay... I've created a quick and dirty example here, https://github.com/pion/mediadevices/tree/pcm16/examples/file.

The APIs are not the most user-friendly. Please let me know if the example is not clear, I can add more comments.

Unfortunately, I don't have a lot of bandwidth to maintain the project due to my full-time work. So, any contributions from the community will be really helpful to make this project healthy! If interested, contact me at lherman.cs@gmail.com or slack me directly on the Gophers channel.

Laky-64, 2021-06-23:
Ok, thanks, I'll try it

Laky-64, 2021-06-25:
This example works with jsfiddle, but with telegram nop, i don't know why, what you need to know why isn't working @lherman-cs , thanks in advance

Laky-64, 2021-06-25:
This is the section where source code of telegram use webrtc

https://github.com/TelegramMessenger/tgcalls/tree/master/tgcalls/group

Laky-64, 2021-06-27:
This is the two working SessionDescription of Javascript Webrtc, this can be helpful for you to know the problem?

LocalDescription
```
v=0
o=- 8373122765090610582 2 IN IP4 127.0.0.1
s=-
t=0 0
a=group:BUNDLE 0
a=msid-semantic: WMS
m=audio 48595 UDP/TLS/RTP/SAVPF 111 103 104 9 102 0 8 106 105 13 110 112 113 126
c=IN IP4 172.30.238.7
a=rtcp:9 IN IP4 0.0.0.0
a=candidate:2757424436 1 udp 2122260223 172.30.238.7 48595 typ host generation 0 network-id 1
a=candidate:559267639 1 udp 2122202367 ::1 42404 typ host generation 0 network-id 3
a=candidate:1510613869 1 udp 2122129151 127.0.0.1 43418 typ host generation 0 network-id 2
a=candidate:3940092356 1 tcp 1518280447 172.30.238.7 57157 typ host tcptype passive generation 0 network-id 1
a=candidate:1876313031 1 tcp 1518222591 ::1 54633 typ host tcptype passive generation 0 network-id 3
a=candidate:344579997 1 tcp 1518149375 127.0.0.1 56701 typ host tcptype passive generation 0 network-id 2
a=ice-ufrag:BeRp
a=ice-pwd:0BhF5wpNQ83xaBrAarHTNkTD
a=ice-options:trickle
a=fingerprint:sha-256 E2:9C:02:05:7C:3B:A9:75:C9:22:77:BB:08:25:EE:E4:82:53:46:CE:20:BB:57:A8:BD:0F:4D:5B:E7:CB:50:9D
a=setup:actpass
a=mid:0
a=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level
a=extmap:2 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time
a=extmap:3 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01
a=extmap:4 urn:ietf:params:rtp-hdrext:sdes:mid
a=extmap:5 urn:ietf:params:rtp-hdrext:sdes:rtp-stream-id
a=extmap:6 urn:ietf:params:rtp-hdrext:sdes:repaired-rtp-stream-id
a=sendrecv
a=msid:- a7d6f191-91cb-44f2-93d9-40ac90267c55
a=rtcp-mux
a=rtpmap:111 opus/48000/2
a=rtcp-fb:111 transport-cc
a=fmtp:111 minptime=10;useinbandfec=1
a=rtpmap:103 ISAC/16000
a=rtpmap:104 ISAC/32000
a=rtpmap:9 G722/8000
a=rtpmap:102 ILBC/8000
a=rtpmap:0 PCMU/8000
a=rtpmap:8 PCMA/8000
a=rtpmap:106 CN/32000
a=rtpmap:105 CN/16000
a=rtpmap:13 CN/8000
a=rtpmap:110 telephone-event/48000
a=rtpmap:112 telephone-event/32000
a=rtpmap:113 telephone-event/16000
a=rtpmap:126 telephone-event/8000
a=ssrc:3550684645 cname:vAEfVjr8bDzmBmog
a=ssrc:3550684645 msid:- a7d6f191-91cb-44f2-93d9-40ac90267c55
a=ssrc:3550684645 mslabel:-
a=ssrc:3550684645 label:a7d6f191-91cb-44f2-93d9-40ac90267c55
```

I create before the localOffer with nil options, then i request to the telegram the params for to do the connection and i get this for make the remoteDescription based on this

```
{
     "transport":{
          "candidates":[
               {
                    "component":1,
                    "foundation":1,
                    "generation":0,
                    "id":"c720d7d10bd934302bad8ea9",
                    "ip":"2001:67c:4e8:f102:7:0:285:703",
                    "network":0,
                    "port":32001,
                    "priority":2130706431,
                    "protocol":"udp",
                    "type":"host"
               },
               {
                    "component":1,
                    "foundation":2,
                    "generation":0,
                    "id":"7c19830c10bd9343061a69f7b",
                    "ip":"91.108.9.115",
                    "network":0,
                    "port":32001,
                    "priority":2130706431,
                    "protocol":"udp",
                    "type":"host"
               }
          ],
          "fingerprints":[
               {
                    "fingerprint":"BB:06:39:03:AF:C3:82:31:5E:16:20:9B:46:48:0F:21:0A:1D:0C:8B:62:31:79:AC:7F:A0:3C:65:2D:3D:84:44",
                    "hash":"sha-256",
                    "setup":"passive"
               }
          ],
          "pwd":"3v5ejlciut5vtcrtk9q4u68jrm",
          "ufrag":"9atfa1f97dpq4m"
     }
}
```

RemoteDescription
```
v=0
o=- 1624820738288 2 IN IP4 127.0.0.1
s=-
t=0 0
a=group:BUNDLE 0
a=msid-semantic: WMS
a=ice-lite
m=audio 32001 RTP/SAVPF 111 126
c=IN IP4 91.108.9.115
a=rtcp:9 IN IP4 0.0.0.0
a=candidate:1 1 udp 2130706431 2001:67c:4e8:f102:7:0:285:703 32001 typ host generation 0
a=candidate:2 1 udp 2130706431 91.108.9.115 32001 typ host generation 0
a=ice-ufrag:9atfa1f97dpq4m
a=ice-pwd:3v5ejlciut5vtcrtk9q4u68jrm
a=fingerprint:sha-256 BB:06:39:03:AF:C3:82:31:5E:16:20:9B:46:48:0F:21:0A:1D:0C:8B:62:31:79:AC:7F:A0:3C:65:2D:3D:84:44
a=setup:passive
a=mid:0
a=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level
a=recvonly
a=rtcp-mux
a=rtpmap:111 opus/48000/2
a=rtcp-fb:111 transport-cc
a=fmtp:111 minptime=10;usedtx=1;useinbandfec=1
a=rtpmap:126 telephone-event/8000
```

Laky-64, 2021-07-02:
I published also a code semplification of what i do
https://vitHub.com/pytgcalls/ntgcalls

Laky-64, 2021-07-02:
I published also a code semplification of what i do
https://github.com/pytgcalls/ntgcalls

@lherman-cs

jech, 2021-07-05:
According to RFC 3551, 16-bit PCM should be described by an a=rtmap line with the codec name `L16`.  I don't see such a line in any of the SDP fragments that you posted.

Please describe accurately what you need.

Laky-64, 2021-07-06:
@jech do you know how to help me to use the following data?

If the problem does not persist in the PCM16L, if you have telegram we can write there

Laky-64, 2021-07-06:
[image]
According to this I don't think I am making any configuration errors, but I think it is more instead a lack of PCM16L codec, and I think it needs PCM16L to transmit, or else there is some strange problem not known to me, because same configuration on Javascript works, but I use PCM16L and not opus

Laky-64, 2021-07-07:
[image]
This is what I want to do, I made a concept map

Link1: https://github.com/pytgcalls/pytgcalls/blob/master/src/stream.ts
Link2: https://github.com/pytgcalls/pytgcalls/blob/master/src/sdp-builder.ts

Laky-64, 2021-07-18:
@Sean-Der I don't want to bother, but do you know how to help me do this (As Herman said he doesn't have much time as he has a full time job)? I gave any necessary means, created the concept map, explained in detail what I needed and I also did all the necessary tests without asking you every single thing every time, but I kindly ask for help, I tried to read everything about WebRTC, but I can understand to some extent about WebRTC.

I await your reply with patience, thanks in advance

at-wat, 2021-07-18:
In my understanding, you want to port https://github.com/pytgcalls/pytgcalls to Golang. Is it right?

In https://github.com/pytgcalls/pytgcalls, raw PCM16L audio is passed to `node-webrtc`'s `RTCAudioSource`.
https://github.com/pytgcalls/pytgcalls/blob/9b181ff53695a99dbe5b003d6f86cfcea426e59c/src/stream.ts#L250-L256
It **doesn't** mean that the WebRTC communication between Telegram peers are using PCM16L as a codec.
(Codec is automatically negotiated and the audio is internally encoded using the negotiated codec.)

According to the SDPs in https://github.com/pion/mediadevices/issues/339#issuecomment-869238310, candidate of the codecs are:
- local: Opus, ISAC, G.722, PCMU, PCMA, iLBC
- remote: Opus

(CN and telephone-event are not audio itself and used to with other codecs)
So, the actually used codec in Telegram should be Opus.

Laky-64, 2021-07-18:
@at-wat  Yes, i want to Port pytgcalls to Golang

while for the version in Nodejs it transmits the audio, the one in go fails, I made the code readable to all:

https://github.com/pytgcalls/ntgcalls

Laky-64, 2021-07-21:
@at-wat 
I tried to use opus as a codec, but it doesn't stream audio, and as stated in https://github.com/TelegramMessenger/tgcalls/blob/master/tgcalls/group/GroupInstanceCustomImpl.cpp#L888-L964 it implies they use PCM16L , since isRawPcm is always true

Laky-64, 2021-07-21:
@at-wat Where can I understand if PCM16L is used as their internal codec for communication between WebRTC and Telegram?
There is an official telegram lib which is used for the Desktop app.
I don't know if this is useful
https://github.com/TelegramMessenger/tgcalls/blob/master/tgcalls/group/GroupInstanceCustomImpl.cpp#L888-L964

roj1512, 2022-01-27:
@ashellunts about the use case: Just read the issue description.

EmrysMyrddin, 2022-08-01:
Hi, does this issue still pending or a solution has been found ?

Should I keep the issue open ?

roj1512, 2022-08-02:
@EmrysMyrddin Yes, you keep it open.