pion/mediadevices #341: When the camera is connected, it returns exit status 3221226356

mondy, 2021-06-14:
### Your environment.
  - Version: github.com/pion/mediadevices v0.2.0
  - OS: Windows10

### What did you do?
Run the following code.
Set `PION_LOG_DEBUG=all`

```go
package main

import (
	"image/jpeg"
	"log"
	"os"

	"github.com/pion/mediadevices"
	_ "github.com/pion/mediadevices/pkg/driver/camera"
)

func main() {
	stream, err := mediadevices.GetUserMedia(mediadevices.MediaStreamConstraints{
		Video: func(constraint *mediadevices.MediaTrackConstraints) {
		},
	})

	if err != nil {
		log.Fatalln(err)
	}

	track := stream.GetVideoTracks()[0]
	videoTrack := track.(*mediadevices.VideoTrack)
	defer videoTrack.Close()

	videoReader := videoTrack.NewReader(false)
	frame, release, _ := videoReader.Read()
	defer release()

	output, _ := os.Create("frame.jpg")
	jpeg.Encode(output, frame, nil)
}
```

### What did you expect?
The camera should be captured and output to `frame.jpg`.

### What happened?

When the camera is connected, the following will be output.
```
exit status 3221226356
```

When the camera is not connected, the following will be output.
```
mediadevices DEBUG: 13:50:58.335573 mediadevices.go:151: 
============ Found Properties ============

=============== Constraints ==============

DeviceID: any
VideoConstraints:
  Width: any
  Height: any
  FrameRate: any
  FrameFormat: any
AudioConstraints:
  ChannelCount: any
  Latency: any
  SampleRate: any
  SampleSize: any
  IsBigEndian: any
  IsFloat: any
  IsInterleaved: any

================ Best Fit ================

Not found
2021/06/14 13:50:58 failed to find the best driver that fits the constraints
exit status 1
```

What does exit status 3221226356 mean?

mondy, 2021-07-01:
Is there any progress on this issue?
Does the mediadevices module work properly? Is it under development?
Can I use any webcam? Are there any restrictions on the specifications of the available webcams?

at-wat, 2021-07-12:
Most of the contributors of this repository are using Linux, so I guess anyone is working on it.

I tried `examples/http` on the commit 070ab924f92a577612841242995aa750897c0e4a (current master) cross compiled by Go 1.16.2 and x86_64-w64-mingw32-gcc 10.3.1.
It works on Windows 11 build 22000.co_release.210604-1628 with [a USB camera](https://www.buffalo.jp/product/detail/bsw20km11bk.html).

Windows version of `pkg/driver/camera` is based on DirectShow. The camera should be compatible with DirectShow.

at-wat, 2021-07-12:
I confirmed Intel RealSense R200 works on the above condition too.

mondy, 2022-03-01:
I got a new USB camera. And fix enumlate windows camera.

### Your environment.

- Go: go1.17.7 windows/amd64
- GCC: mingw-w64-x86_64-gcc 11.2.0-8
- MediaDevices: github.com/pion/mediadevices master(e316b30) + fix
- OS: Windows 10

### Fix details

I forked it and fixed [it here](https://github.com/mondy/mediadevices/tree/fix-enumlate-windows-camera). A few fix.
I do not understand DirectShow.

Look at [the commit](https://github.com/mondy/mediadevices/commit/a792eca3249e4767ea7f5b9bb1048367ca9248b0).

The first fix is to forget to increment the index. This is a general bug that does not involve DirectShow, so it should not be a problem.

The second fix is to fail if the captureFilter is nullptr. Fixed an error that occurred when trying to access captureFilter in the getPin function. I don't know if this fix is correct because I don't understand DirectShow.

I have only confirmed that the program that outputs the image above works.

In the meantime, I have identified the bug and forked it to report it.
Please let me know if you would like me to pull request for you.

EmrysMyrddin, 2022-08-01:
If you still have the occasion to open a PR for this, it would be awesome !