pion/mediadevices #342: AVFoundationBind error when vendored

Antonito, 2021-06-20:
### Your environment.
  - Version: v0.2.0
  - Browser: CLI

### What did you do?

```
# Here, there's no issue, everything works as expected
rm -rf vendor/
go run .

# However if vendored, it doesn't
go mod vendor
go run .

# github.com/pion/mediadevices/pkg/codec/opus
./vendor/github.com/pion/mediadevices/pkg/codec/opus/opus.go:15:10: fatal error: 'opus.h' file not found
#include <opus.h>
         ^~~~~~~~
1 error generated.
# github.com/pion/mediadevices/pkg/avfoundation
./vendor/github.com/pion/mediadevices/pkg/avfoundation/avfoundation_darwin.go:6:11: fatal error: 'AVFoundationBind/AVFoundationBind.h' file not found
 #include "AVFoundationBind/AVFoundationBind.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
```

Here is the code I'm using – inspired by the `webrtc` example.

```go
func setupHardwareTracks() (mediadevices.Track, mediadevices.Track) {
	x264Params, err := x264.NewParams()
	if err != nil {
		panic(err)
	}

	x264Params.Preset = x264.PresetMedium
	x264Params.BitRate = 1_000_000 // 1mbps

	opusParams, err := opus.NewParams()
	if err != nil {
		panic(err)
	}
	codecSelector := mediadevices.NewCodecSelector(
		mediadevices.WithVideoEncoders(&x264Params),
		mediadevices.WithAudioEncoders(&opusParams),
	)

	s, err := mediadevices.GetUserMedia(mediadevices.MediaStreamConstraints{
		Video: func(c *mediadevices.MediaTrackConstraints) {
			c.FrameFormat = prop.FrameFormat(frame.FormatI420)
			c.Width = prop.Int(640)
			c.Height = prop.Int(480)
		},
		Audio: func(c *mediadevices.MediaTrackConstraints) {
		},
		Codec: codecSelector,
	})
	if err != nil {
		panic(err)
	}

	return s.GetVideoTracks()[0], s.GetAudioTracks()[0]
}
```

### What did you expect?

The code should be able to run after a `go mod vendor`

Antonito, 2021-06-29:
Might be related to https://github.com/pion/mediadevices/issues/290

EmrysMyrddin, 2022-08-01:
Can you check for your XCode version ? As stated in the issue you linked.