pion/mediadevices #347: Windows 10 amd64 can not open Camera

fjfo, 2021-07-14:
### Your environment.
  - Version: golang 1.16
  - OS: Windows 10, amd64
### What did you do?
        x264Params, err := x264.NewParams()
	must(err)
	x264Params.Preset = x264.PresetMedium
	x264Params.BitRate = 1_000_000 // 1mbps

	codecSelector := mediadevices.NewCodecSelector(
		mediadevices.WithVideoEncoders(&x264Params),
	)

	mediaStream, err := mediadevices.GetUserMedia(mediadevices.MediaStreamConstraints{
		Video: func(c *mediadevices.MediaTrackConstraints) {
			c.FrameFormat = prop.FrameFormat(frame.FormatI420)
			c.Width = prop.Int(640)
			c.Height = prop.Int(480)
		},
		Codec: codecSelector,
	})
	must(err)

	videoTrack := mediaStream.GetVideoTracks()[0].(*mediadevices.VideoTrack)
	defer videoTrack.Close()
	videoReader := videoTrack.NewReader(false)
	var buf bytes.Buffer

		imgFrame, release, err := videoReader.Read()
		if err == io.EOF {
			return
		}
		must(err)

		out, _ := os.Create("./img.jpeg")
		defer out.Close()

		err = jpeg.Encode(out, imgFrame, nil)
		if err != nil {
			log.Println(err)
		}
		release()
		buf.Reset()

### What did you expect?
- Open Camera and write 1 Frame to disk
### What happened?
- Error: errConnectFilters = "failed to connect filters";

fjfo, 2021-07-14:
More debug log here

`# github.com/pion/mediadevices/pkg/driver/camera
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: $WORK\b184\_x002.o: in function `_cgo_a5197d6edf98_Cfunc_getName':
/tmp/go-build/cgo-gcc-prolog:119: undefined reference to `getName'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: $WORK\b184\_x002.o: in function `_cgo_a5197d6edf98_Cfunc_getProp':
/tmp/go-build/cgo-gcc-prolog:139: undefined reference to `getProp'
collect2.exe: error: ld returned 1 exit status`

fjfo, 2021-07-14:
It seems 

"github.com/pion/mediadevices/pkg/driver/camera" // This is required to register camera adapter

Can not work in Windows 10????????

Any helps please!

BenLocal, 2022-04-25:
same issues,are there any gcc dependencies missing?

```
C:/TDM-GCC-64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: $WORK\b190\_x002.o: in function `_cgo_3c948bc5c0b5_Cfunc_getName':
/tmp/go-build/cgo-gcc-prolog:119: undefined reference to `getName'
C:/TDM-GCC-64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: $WORK\b190\_x002.o: in function `_cgo_3c948bc5c0b5_Cfunc_getProp':
/tmp/go-build/cgo-gcc-prolog:139: undefined reference to `getProp'
collect2.exe: error: ld returned 1 exit status (exit status 2)
```

myahuang, 2022-08-21:
I copy the snippet example ,and run, then find same issue.The program have find one camera,but could't run.`	stream, err := mediadevices.GetUserMedia(mediadevices.MediaStreamConstraints{
		Video: func(constraint *mediadevices.MediaTrackConstraints) {
			// Query for ideal resolutions
			constraint.Width = prop.Int(600)
			constraint.Height = prop.Int(400)
		},
	})
	fmt.Println("stream error", err)`
the terminal message is: stream error failed to open device: failed to connect filters

qiulin, 2024-03-16:
You may need to install CCCP, https://answers.informer.com/24666/receive-combination-intermediate-filters-found-connection.

It works for me.

closed 2024-04-29 by edaniels