pion/mediadevices #359: unresolvable R_AARCH64_ADR_PREL_PG_HI21 relocation against symbol `stderr@@GLIBC_2.17'

PruthvinathR, 2021-09-23:
I get this error when I build mediadevices on arm64.

The glibc version on my arm64 is 2.27. Can that be the reason for the error?

PruthvinathR, 2021-09-24:
@lherman-cs Do you think the issue is because the opus library on arm64 is built against glibc 2.17? If so can you please commit the files in the scripts folder so that we can build it ourselves.

at-wat, 2021-09-24:
@PruthvinathR I would recommend to use system-installed library if a package manager providing opus is available on your environment.
Setting `-tags dynamic` makes mediadevices to use codec libraries found by pkg-config instead of the bundled ones.

PruthvinathR, 2021-09-24:
@at-wat thanks for your comment. I didn't know how to build the way you mentioned. 

What I did was, I built opus on arm64 and replaced the libopus-linux-arm64.a with the libopus.a that I built. It works now.

I think instead of including the prebuilt libopus-linux-arm64.a in mediadevices, it will much better if we can add the libopus build step to the mediadevices installation process.

closed 2021-09-27 by PruthvinathR