pion/mediadevices #374: how to play the sound from  the wave.Audio

hongshengjie, 2021-12-28:
`
trackaudio := mediaStream.GetAudioTracks()[0]
audioTrack := trackaudio.(*mediadevices.AudioTrack)
audioReader := audioTrack.NewReader(false)
chunk, release, err := audioReader.Read()
`
how to use `chunk` to play the sound?

EmrysMyrddin, 2022-08-01:
What do you want to do ?

The chunk returned by `Read` is a chunk of audio meant to be then sent or saved to a file. To have a full audio record/stream, you have to call `Read` in a loop and constantly reading the audio chunks.

Whatever you do with the chunks, keep in mind that the treatment has to be quicker than the frequence of chunk production of the audioReader. Otherwise, you will miss some chunk, resulting in audio artefacts/missing parts.

EmrysMyrddin, 2022-08-01:
I will close this issue, but if you still have questions, don't hesitate to reopen this issue or a new one with your new question

closed 2022-08-01 by EmrysMyrddin