pion/mediadevices #379: "panic: send on closed channel" after call Close on peerConnection

migueldevmt, 2022-02-22:
### Your environment.
  - Version: github.com/pion/mediadevices v0.3.2, github.com/pion/webrtc/v3 v3.1.10
  - Browser: Chrome 98.0.4
  - Other Information:
####  Stacktrace:

> panic: send on closed channel
> 
> goroutine 369 [running]:
> github.com/pion/mediadevices.(*baseTrack).unbind(0x3962393261333461, {{0xc00027c8d0, 0x20}, {{0x7ff693bfa658, 0x0, 0x0}, {0xc000f00000, 0x1, 0x1}}, 0xec9d1d4, ...})
> 	$GOPATH/pkg/mod/github.com/pion/mediadevices@v0.3.2/track.go:227 +0x70
> github.com/pion/mediadevices.(*VideoTrack).Unbind(0x0, {{0xc00027c8d0, 0x20}, {{0x7ff693bfa658, 0x0, 0x0}, {0xc000f00000, 0x1, 0x1}}, 0xec9d1d4, ...})
> 	$GOPATH/pkg/mod/github.com/pion/mediadevices@v0.3.2/track.go:311 +0x45
> github.com/pion/webrtc/v3.(*RTPSender).ReplaceTrack(0xc00016c2c0, {0x0, 0x0})
> 	$GOPATH/pkg/mod/github.com/pion/webrtc/v3@v3.1.10/rtpsender.go:157 +0x198
> github.com/pion/webrtc/v3.(*RTPSender).Stop(0xc00016c2c0)
> 	$GOPATH/pkg/mod/github.com/pion/webrtc/v3@v3.1.10/rtpsender.go:240 +0x85
> github.com/pion/webrtc/v3.(*RTPTransceiver).Stop(0xc00019a510)
> 	$GOPATH/pkg/mod/github.com/pion/webrtc/v3@v3.1.10/rtptransceiver.go:148 +0x7e
> github.com/pion/webrtc/v3.(*PeerConnection).Close(0xc000029200)
> 	$GOPATH/pkg/mod/github.com/pion/webrtc/v3@v3.1.10/peerconnection.go:1878 +0x1c5
> gitlab.com/awati/vrp-server/peerconnection.CloseConnection(0xc0000a3fd0, 0x23c)
> 	$PROJECT_FOLDER/peerconnection/peerconnection.go:360 +0x148
> gitlab.com/awati/vrp-server/peerconnection.StartEngine.func1(0x5)
> 	$PROJECT_FOLDER/peerconnection/peerconnection.go:117 +0x48
> created by github.com/pion/webrtc/v3.(*PeerConnection).onConnectionStateChange
> 	$GOPATH/pkg/mod/github.com/pion/webrtc/v3@v3.1.10/peerconnection.go:488 +0x175

### What did you do?
I call a close function when the state changes to some imcompatible status:
> peerConnection.OnConnectionStateChange(func(pcs webrtc.PeerConnectionState) {
> 	switch pcs {
> 	case webrtc.PeerConnectionStateDisconnected, webrtc.PeerConnectionStateClosed, webrtc.PeerConnectionStateFailed:
> 		CloseConnection(peerConnection, pid)
> 		fallthrough
> 		...
and inside the CloseConnection i call peerConnection.Close()

### What did you expect?
To close the connection and continue to run the program
 
### What happened?
the process end with a panic message

Meonardo, 2022-03-31:
have the same problem when I call `CloseConnection .Close()`, and this happend occasionilly.

Hevienz, 2022-04-21:
Same problem, any one can help?

EmrysMyrddin, 2022-07-25:
This happens when you try to close a peer connection or a sender associated to a media track already closed.

I didn't found a real workaround, but made a PR which should hopefully be merged quickly.

closed 2022-07-27 by EmrysMyrddin

Meonardo, 2022-09-01:
New problem: "panic: close of nil channel"

Step to reproduce:
Call `PeerConnection.Close()`

---
OS: Windows11(64)
Go: 1.17

---

Please see the callstack:
[image]

PR #414 (2022-07-27): Fix closing a camera not closing the associated track
#### Description

Closing a record session on macOS does not close the reader.

This PR fixes this by closing the channel used to read incoming video frames directly in the Close function.
Calling it in the data callback is not working because since the record session is closed, the callback will never be called again. So the reader will indefinitely wait for a incoming frame/the channel to be closed, which will never happen.

By fixing this, i have also discover a bug, where closing a peer connection with some tracks already closed trigger [... cut]