pion/mediadevices #391: Cant run/build homepage example on RaspberryPi

jonathongardner, 2022-03-30:
### Your environment.
  - Version: commit: 70261260cb6ff658d20d43c6fd0145c8f6d24811 
  - Other Information - raspbian OS Lite 

### What did you do?
Tried running the example on the homepage:
```go
package main

import (
	"image/jpeg"
	"os"

	"github.com/pion/mediadevices"
	"github.com/pion/mediadevices/pkg/prop"

	// This is required to register camera adapter
	_ "github.com/pion/mediadevices/pkg/driver/camera" 
	// Note: If you don't have a camera or your adapters are not supported,
	//       you can always swap your adapters with our dummy adapters below.
	// _ "github.com/pion/mediadevices/pkg/driver/videotest"
)

func main() {
	stream, _ := mediadevices.GetUserMedia(mediadevices.MediaStreamConstraints{
		Video: func(constraint *mediadevices.MediaTrackConstraints) {
			// Query for ideal resolutions
			constraint.Width = prop.Int(600)
			constraint.Height = prop.Int(400)
		},
	})

	// Since track can represent audio as well, we need to cast it to 
	// *mediadevices.VideoTrack to get video specific functionalities
	track := stream.GetVideoTracks()[0]
	videoTrack := track.(*mediadevices.VideoTrack)
	defer videoTrack.Close()

	// Create a new video reader to get the decoded frames. Release is used 
	// to return the buffer to hold frame back to the source so that the buffer 
	// can be reused for the next frames.
	videoReader := videoTrack.NewReader(false)
	frame, release, _ := videoReader.Read()
	defer release()

	// Since frame is the standard image.Image, it's compatible with Go standard 
	// library. For example, capturing the first frame and store it as a jpeg image.
	output, _ := os.Create("frame.jpg")
	jpeg.Encode(output, frame, nil)
}
```
than
```sh
go run main.go
```

### What did you expect?
To capture a frame and save it.

### What happened?
```
panic: runtime error: invalid memory address or nil pointer dereference
...
```

I tried a couple of other things but couldn't figure anything out on the raspberry pi

jonathongardner, 2022-04-06:
Alright, I was able to figure it out. I am using the most recent Raspberry Pi OS so I had to enable the legacy camera (I probably should have noticed that by how old the benchmark is/using mmal). Not sure if most users would know that but might be nice to have that [here](https://github.com/pion/mediadevices#mmal). If most people would i guess this could be closed.

jason-shen, 2022-04-09:
just a question here @jonathongardner is there any effort getting mic working on a pi? or does it just works

EmrysMyrddin, 2022-08-01:
Closing in favor of #407

closed 2022-08-01 by EmrysMyrddin