pion/mediadevices #397: Implement custom mediadevice

Coimbra1984, 2022-04-08:
## Summary

Possibility to implement a custom media device

## Motivation

I have an RTSP camera here which I want to bridge to webrtc. Currently I use https://github.com/aler9/gortsplib as a client library to receive the camera stream and push the data through webrtc using pion/webrtc (PeerConnection). Unfortunately as far as I understand, there is also an RTCP channel, where I receive incoming messages like NACK, PLI and other packages. As far as I understand, this mediadevices library already also handles these messages correctly but only work for input devices on the PC. Is there a possibility to implement my own mediadevice as described above?

EmrysMyrddin, 2022-07-25:
It is already possible to create your own [`Adapter`](https://github.com/pion/mediadevices/blob/d038133783323061ce3208afaf1bad2f5835739b/pkg/driver/driver.go#L35) and register a device using this adapter using [`driver.Manager.Register`](https://github.com/pion/mediadevices/blob/d038133783323061ce3208afaf1bad2f5835739b/pkg/driver/manager.go#L71).

You can find example how to implement your adapter and register your device by looking to already implemented ones in the [`pkg/driver`](https://github.com/pion/mediadevices/tree/master/pkg/driver) folder. For example using video, take a look to [`screen`] (https://github.com/pion/mediadevices/blob/master/pkg/driver/screen/screen.go) or [`camera`](https://github.com/pion/mediadevices/blob/master/pkg/driver/camera/camera_linux.go).

EmrysMyrddin, 2022-07-29:
I close this issue assuming your question has been answered.

If it's not the case and you still have question about this, don't hesitate to reopen or open a new issue.

closed 2022-07-29 by EmrysMyrddin