pion/mediadevices #424: Optimise ToI420 conversion by using sync.Pool

EmrysMyrddin, 2022-08-01:
Since #418, the I420 conversion have been slow down because 2 new buffers are allocated for each frame converted.

The idea is to use a `sync.Pool` to minimise the overhead of creating new `[]byte`: 
 - we get a `[]byte` with `Pool.Get`, 
 - we check if it's size feats the image to convert, growing it if necessary
 - use it for the conversion
 - return the Image + a release function which calls the `Pool.Put` function with the used `[]byte` buffers.

This way we should limit the garbage collector pressure and the time needed to fill the `[]byte` with 0 values at creation.

neversi, 2023-02-18:
Could I have a try to resolve this issue?

neversi, 2023-02-19:
Is there need to add sync.Pool? Whenever there is the call to ReaderFunc, there is also mutex Lock, so here we could use closure instead as there is no concurrent access to the buffers.

at-wat, 2023-04-24:
done by https://github.com/pion/mediadevices/pull/473

closed 2023-04-24 by at-wat