pion/mediadevices #438: Vaapi not working with webrtc

EmrysMyrddin, 2022-08-16:
### Your environment.
  - Version: v0.3.10
  - Browser: none

### What did you do?
Tried to use the VAAPI binding. Seems like broken, but i don't have any clue on how to fix it.

Here are the logs when trying to publish the track with webrtc (using VP8, but it's not working either with VP9) :
```2022/08/16 14:03:12 "msg"="could not set remote description" "error"="video/VP8: video/VP8: libva entrypoint not found\n\nvideo/H264: video/VP8: %!s(<nil>)\n\nvideo/H264: video/VP8: %!s(<nil>)\n\nvideo/H264: video/VP8: %!s(<nil>)"```

ZeoWorks, 2022-10-07:
VAAPI not working for me either

at-wat, 2022-10-12:
`libva entrypoint not found` usually means the video card doesn't support specified profile.
You can list supported profiles by `vainfo` command and `VAEntrypointEncSlice` for desired codec is needed.

ZeoWorks, 2022-10-13:
> `libva entrypoint not found` usually means the video card doesn't support specified profile. You can list supported profiles by `vainfo` command and `VAEntrypointEncSlice` for desired codec is needed.

Hi, both VP8 & VP9 encoding is supported on my machine.
However VAAPI is simply not working on webrtc.
[image]

at-wat, 2022-10-14:
`VAEntrypointVLD` is for decoding.
`VAEntrypointEncSlice` is required for encoding.

ZeoWorks, 2022-10-14:
> `VAEntrypointVLD` is for decoding. `VAEntrypointEncSlice` is required for encoding.

That's what I thought also.
I'm really surprised the RX 6700 XT does NOT support VPX hardware encoding.
It's a fairly new & expensive card.

at-wat, 2022-10-14:
> Will H264/265 encoding be supported in the future with VAAPI?

VAAPI library supports H264/265 and your hardware supports both. (EncSlice for H264 and HEVC)
Need to add wrappers to this package.

ZeoWorks, 2022-10-14:
> > Will H264/265 encoding be supported in the future with VAAPI?
> 
> VAAPI library supports H264/265 and your hardware supports both. (EncSlice for H264 and HEVC) Need to add wrappers to this package.

Any eta when H264/265 vaapi support?

at-wat, 2022-10-14:
> Any eta when H264/265 vaapi support?

No, as this is an open source project, contributions are always welcomed.