pion/mediadevices #441: reading github.com/pion/mediadevices/go.mod at revision v0.0.0: unknown revision v0.0.0

ZeoWorks, 2022-09-03:
Hi, upon requesting signal;
`go get github.com/pion/mediadevices/examples/internal/signal`

This error occurs;
`go: github.com/pion/mediadevices/examples@v0.0.0-20220824115655-3bec69bbf884 requires
	github.com/pion/mediadevices@v0.0.0: reading github.com/pion/mediadevices/go.mod at revision v0.0.0: unknown revision v0.0.0
`

No issue with any of the others. :)

at-wat, 2022-09-04:
Hi, thanks for trying out the package!
Subpackages under `./examples` aren't designed to be used from other packages or product apps.
(Path including `./internal` means unexported: https://go.dev/doc/go1.4#internalpackages)

If building any product apps, we recommend to implement proper and secure way to exchange SDPs.
If just trying the package, please copy the files under your package.

BTW, the direct reason of the error is due to `replace`:
https://github.com/pion/mediadevices/blob/3bec69bbf8841be66004021d4e8453cf520ab51c/examples/go.mod#L11
This is for separating deps only for examples from the main library and keep the examples buildable with the latest library.

ZeoWorks, 2022-09-07:
Thank you, but this occurs even within your examples.
An example would be trying to build your webrtc sample;
https://github.com/pion/mediadevices/tree/master/examples/webrtc

at-wat, 2022-09-07:
@ZeoWorks Hmm, I couldn't reproduce the error on building `./examples/webrtc` on Go 1.19 on Linux.
What environment and Go version do you use? Also, could you share the exact command and error log?

ZeoWorks, 2022-09-16:
Hi, go.1.19 on Windows, latest mediadevices version. 
Thanks sir!