pion/mediadevices #452: What is wrong with the C interface for opus  opus_encoder_create  ?

trump-7575, 2022-11-01:
### Your environment.
  - Version:  0.3.0
  - OS : Ubuntu 22.04 , amd64 
  - Browser: None ( I did not use browser to run this test )
  - ion-sdk-go version :  latest
  - libopus-dev:  1.3.1-0.1build2

### What did you do?
I modified the official example :  example/ion-sfu-join-from-mediadevice/main.go  at  ion-sdk-go  respository ( https://github.com/ionorg/ion-sdk-go ) to try to publish audio ( opus ) only to sfu.

The modificaion : 
=========== original ==============

	vpxParams, err := vpx.NewVP8Params()
	if err != nil {
		panic(err)
	}
	vpxParams.BitRate = 500_000 // 500kbps

	codecSelector := mediadevices.NewCodecSelector(
		mediadevices.WithVideoEncoders(&vpxParams),
	)

	fmt.Println(mediadevices.EnumerateDevices())

	s, err := mediadevices.GetUserMedia(mediadevices.MediaStreamConstraints{
		Video: func(c *mediadevices.MediaTrackConstraints) {
			c.FrameFormat = prop.FrameFormat(frame.FormatYUY2)
			c.Width = prop.Int(640)
			c.Height = prop.Int(480)
		},
		Codec: codecSelector,
	})
============ My modification =================

	opusParams, err := opus.NewParams()
	if err != nil {
		panic(err)
	}
	
	codecSelector := mediadevices.NewCodecSelector(
		mediadevices.WithAudioEncoders(&opusParams),
	)
	

	fmt.Println(mediadevices.EnumerateDevices())

	s, err := mediadevices.GetUserMedia(mediadevices.MediaStreamConstraints{
		Audio: func(c *mediadevices.MediaTrackConstraints) {
		},
		Codec: codecSelector,
	})

### What did you expect?

Audio track is published to sfu.

### What happened?
Failed to create the opus encoder when calling the C function    :    

 parsed:<nil>} 
SIGILL: illegal instruction
PC=0xaa89a1 m=5 sigcode=2
signal arrived during cgo execution
instruction bytes: 0xc5 0xf9 0x6f 0x5 0x47 0xa8 0x2c 0x0 0x48 0x83 0xe7 0xf8 0x31 0xc0 0x48 0x29

goroutine 53 [syscall]:
runtime.cgocall(0xa237c0, 0xc0000bc930)
	/usr/local/go/src/runtime/cgocall.go:158 +0x5c fp=0xc0000bc908 sp=0xc0000bc8d0 pc=0x4134fc
github.com/pion/mediadevices/pkg/codec/opus._Cfunc_opus_encoder_create(0xbb80, 0x2, 0x800, 0xc000323360)
	_cgo_gotypes.go:118 +0x4d fp=0xc0000bc930 sp=0xc0000bc908 pc=0xa1c42d
github.com/pion/mediadevices/pkg/codec/opus.newEncoder({0xd3a9c0, 0xc00032a828}, {{0x0, 0x0}, {0x0, 0x0, 0x0, {0x0, 0x0}}, {0x2, ...}}, ...)
	/home/trump/go/pkg/mod/github.com/pion/mediadevices@v0.3.0/pkg/codec/opus/opus.go:51 +0x149 fp=0xc0000bc9c8 sp=0xc0000bc930 pc=0xa1c709
github.com/pion/mediadevices/pkg/codec/opus.(*Params).BuildAudioEncoder(0x0?, {0xd3a9c0?, 0xc00032a828?}, {{0x0, 0x0}, {0x0, 0x0, 0x0, {0x0, 0x0}}, ...})
	/home/trump/go/pkg/mod/github.com/pion/mediadevices@v0.3.0/pkg/codec/opus/params.go:71 +0xaf fp=0xc0000bca98 sp=0xc0000bc9c8 pc=0xa1bfef
github.com/pion/mediadevices.(*CodecSelector).selectAudioCodecByNames(0xc00039ae40, {0xd3a9c0, 0xc00032a828}, {{0x0, 0x0}, {0x0, 0x0, 0x0, {0x0, 0x0}}, ...}, ...)
	/home/trump/go/pkg/mod/github.com/pion/mediadevices@v0.3.0/codec.go:115 +0x439 fp=0xc0000bcbe0 sp=0xc0000bca98 pc=0xa14599
github.com/pion/mediadevices.(*AudioTrack).newEncodedReader(0xc000300450, {0xc0000bcde8, 0x1, 0x1})
	/home/trump/go/pkg/mod/github.com/pion/mediadevices@v0.3.0/track.go:434 +0x165 fp=0xc0000bcd70 sp=0xc0000bcbe0 pc=0xa1a4c5
github.com/pion/mediadevices.(*AudioTrack).NewRTPReader(0xc000300450, {0xc000322fe6?, 0xc000320270?}, 0x8be88d61, 0x4b0)
	/home/trump/go/pkg/mod/github.com/pion/mediadevices@v0.3.0/track.go:468 +0x6a fp=0xc0000bce08 sp=0xc0000bcd70 pc=0xa1a8ea

-------------------------------------------------
If you need more info. , please do not hesitate to let me know. I do not want to debug an outdate version of mediadevices either. If any of you have a working demo to publish audio to sfu via ion-sdk-go  , please kindly give me an example. Thanks !

trump-7575, 2022-11-01:
go version : go1.19.2 linux/amd64

at-wat, 2022-11-02:
`illegal instruction` on calling external library function is usually caused when the linked library is not supported on the environment. Please check the library your binary linked is correct.