pion/mediadevices #475: Failed to start streaming: invalid argument

johnscode, 2023-02-23:
### Your environment.
  - Version: v0.40
  - Browser: none
  - Other Information - pi running ubuntu 22.10 using an onboard camera (not usb cam)

### What did you do?
Trying to run the jpeg snapshot example from the main Readme:
import (
	"image/jpeg"
	"os"

	"github.com/pion/mediadevices"
	"github.com/pion/mediadevices/pkg/prop"

	// This is required to register camera adapter
	_ "github.com/pion/mediadevices/pkg/driver/camera" 
	// Note: If you don't have a camera or your adapters are not supported,
	//       you can always swap your adapters with our dummy adapters below.
	// _ "github.com/pion/mediadevices/pkg/driver/videotest"
)
func main() {
	stream, _ := mediadevices.GetUserMedia(mediadevices.MediaStreamConstraints{
		Video: func(constraint *mediadevices.MediaTrackConstraints) {
			// Query for ideal resolutions
			constraint.Width = prop.Int(640)
			constraint.Height = prop.Int(480)
		},
	})

	// Since track can represent audio as well, we need to cast it to
	// *mediadevices.VideoTrack to get video specific functionalities
	track := stream.GetVideoTracks()[0]
	videoTrack := track.(*mediadevices.VideoTrack)
	defer videoTrack.Close()

	// Create a new video reader to get the decoded frames. Release is used
	// to return the buffer to hold frame back to the source so that the buffer
	// can be reused for the next frames.
	videoReader := videoTrack.NewReader(false)
	frame, release, _ := videoReader.Read()
	defer release()

	// Since frame is the standard image.Image, it's compatible with Go standard
	// library. For example, capturing the first frame and store it as a jpeg image.
	output, _ := os.Create("frame.jpg")
	jpeg.Encode(output, frame, nil)
}
example runs fine on OSX. runs to completion on pi when substituting the test drivers in the imports, ie
//_ "github.com/pion/mediadevices/pkg/driver/camera"
_ "github.com/pion/mediadevices/pkg/driver/videotest"

attempt to run using
go run main.go

### What did you expect?
successful run producing a jpeg file

### What happened?
 panic failure at line 28

error is occurring due to the line:
stream, _ := mediadevices.GetUserMedia( ....

If i change it to check the error:
stream, err := mediadevices.GetUserMedia( ....
if err != nil {
  fmt.Printf("%+v\n",err)
  return
}
error reported is: 'Failed to start streaming: invalid argument'

If I run (with the real driver imported) using: export PION_LOG_DEBUG=all && go run main.go
i get output showing several configurations that match the constraints, with final one shown being 'best fit':

================ Best Fit ================

DeviceID: b1a30cff-6365-4313-9c8f-e96fd5cadee0
Video:
  Width: 640
  Height: 480
  FrameRate: 0
  FrameFormat: UYVY
Audio:
  ChannelCount: 0
  Latency: 0s
  SampleRate: 0
  SampleSize: 0
  IsBigEndian: false
  IsFloat: false
  IsInterleaved: false

then the error

Should point out that this is using an onboard pi camera and not a usb camera.

fwiw: 
the onboard camera is accessible using the standard libcamera apis (in cpp). 
the blackjack webcam lib works for usb cams on a pi, but not the onboard cam

malykhin, 2023-02-25:
Hey @johnscode,
Is legacy camera support enabled? (https://www.youtube.com/watch?v=E7KPSc_Xr24)

johnscode, 2023-02-25:
yes. using raspi-config, legacy camera support was enabled for this.

florianbgt, 2023-04-08:
I have the same issue trying to run the `examples/http` using a raspberry pi 4 with a pi camera 3

cleroux, 2023-04-24:
I've also just started getting this error.  I suspect underlying Pi camera driver must have automatically updated because my Go service was working with the camera stream until my Pi was rebooted a couple days ago.  The camera stream works fine on my laptop so I think this issue is probably at a lower level than Pion. I will report any further findings here.

jamespilgrim, 2024-04-29:
FWIW, I recently updated libcamera (and rpicam-app), and now I get:
[5:14:13.575871523] [8761] ERROR V4L2 v4l2_videodevice.cpp:1906 /dev/video0[21:cap]: Failed to start streaming: Invalid argument

whenever I've changed the cropping width/height away from the default.  For example,
 media-ctl -d "/dev/media3" --set-v4l2 "'imx296 10-001a':0 [fmt:SBGGR10_1X10/100x100 crop:(0,0)/100x100]" -v

Has anyone had a similar experience?