pion/mediadevices #477: go build linux error

anamyip, 2023-03-01:
**I tried to build the example of examples/webrtc into Linux in the Mac environment, but an error occurred.
What is the reason and how to solve it?**

**Compile webrtc example**

`CGO_ENABLED=0 GOOS=linux GOARCH=amd64 go build main.go `

**The following error occurs**

```
# github.com/pion/mediadevices/pkg/driver/microphone
../../pkg/driver/microphone/microphone.go:29:16: undefined: malgo.AllocatedContext
../../pkg/driver/microphone/microphone.go:36:8: undefined: malgo.DeviceInfo
../../pkg/driver/microphone/microphone.go:48:19: undefined: malgo.InitContext
../../pkg/driver/microphone/microphone.go:48:42: undefined: malgo.ContextConfig
../../pkg/driver/microphone/microphone.go:86:31: undefined: malgo.DeviceInfo
../../pkg/driver/microphone/microphone.go:105:19: undefined: malgo.DeviceConfig
../../pkg/driver/microphone/microphone.go:106:22: undefined: malgo.DeviceCallbacks
../../pkg/driver/microphone/microphone.go:123:30: m.ID undefined (type *microphone has no field or method ID)
../../pkg/driver/microphone/microphone.go:141:23: undefined: malgo.InitDevice
../../pkg/driver/microphone/microphone.go:199:27: m.Formats undefined (type *microphone has no field or method Formats)
../../pkg/driver/microphone/microphone.go:199:27: too many errors
# github.com/pion/mediadevices/pkg/codec/x264
../../pkg/codec/x264/params.go:49:9: undefined: newEncoder
# github.com/pion/mediadevices/pkg/codec/opus
../../pkg/codec/opus/params.go:71:9: undefined: newEncoder
```

at-wat, 2023-03-01:
Hi @anamyip, these codecs requires CGO since they use C libraries.

anamyip, 2023-03-01:
> Hi @anamyip, codecs requires CGO since they use C libraries.

How to modify the following command?

`CGO_ENABLED=0 GOOS=linux GOARCH=amd64 go build main.go `