pion/mediadevices #492: Video not showing in video element in latest versions of chrome.

anand-dotworld, 2023-04-03:
I am trying out the pion mediadevices example for webrtc. which is given in this repository. The connection is connected successfully. I am also receiving a video track and attaching the Mediastream from the video track to a video element, just like in the example I haven't changed anything. But the video is not showing in video element. 

Tested in
Chrome 111 - Not working
Chrome 108 - Working.

I don't know whether the problem is with pion or chrome. Does anyone has similar problem like this. Can anyone help me with this?

naveenrobo, 2023-04-04:
Yes. Unable to steam camera using given webrtc example.

selvan, 2023-04-05:
same issue with chrome 112 and webrtc example. After replacing video encoding H264 with VP8 (changing ```x264.NewParams()``` to ```vpx.NewVP8Params()```), video is showing in Chrome 112.

naveenrobo, 2023-04-07:
Yes. I can confirm vp8 works fine but not h264.

G2G2G2G, 2023-05-04:
every example is broken in 1 or more ways except mjpeg (http) one