pion/mediadevices #500: Distorted audio capture on linux.

dezi, 2023-05-07:
The problem is in pkg/driver/microphone/microphone.go lines 133...

	onRecvChunk := func(_, chunk []byte, framecount uint32) {
		select {
		case <-cancelCtx.Done():
		case m.chunkChan <- chunk:
		}
	}

Callback function of malgo audio capture. The chunk buffer is send to channel.
Malgo will reuse the buffer while it is beeing processed.

You have to make a copy of chunk before sending it to the channel.

	onRecvChunk := func(_, chunk []byte, framecount uint32) {
		newChunk := make([]byte, len(chunk))
		copy(newChunk, chunk)
		select {
		case <-cancelCtx.Done():
		case m.chunkChan <- newChunk:
		}
	}

Please fix.
Regards,
dezi

schrockwell, 2023-11-03:
I have confirmed that this does partially fix the microphone on the Raspberry Pi 4. Where there was once a lot of distortion, there is now none!

But alas, the audio completely stops after a few seconds.

kadharsh, 2023-11-24:
I faced the same problem of severely distorted audio, and this solution worked for me.
Initially, ALSA wasn't providing any audio output; although I could record audio using `arecord`, it wasn't being sent to the other side. After installing PulseAudio, the audio started working, but the quality remained heavily distorted. This fix by @dezi  effectively resolved the distortion issue.

Lien03, 2023-12-22:
I am also experiencing issues with distorted audio,
this worked for me.
- linux-imx 4.14.98
- USB Mic
- PulseAudio

thx

suosuo6, 2024-01-28:
@dezi Hi, I don't quite understand why copying a new byte can solve this problem. Can you help me explain? thanks

kenzoi, 2024-02-22:
I had the same problem when using my bluetooth earbud some months ago, copy the buffer worked at that time.
But now it's working fine for me without any patch, I don't know if it was an update on my Fedora or on pion/mediadevices.

dezi, 2024-04-03:
[schrockwell](https://github.com/schrockwell): I have also experienced the 

[ALSA] poll() failed.

Bug. 

You will find my solution here:

https://github.com/mackron/miniaudio/issues/836

Regards, dezi