pion/mediadevices #545: use mediadevices.GetUserMedia() failed. err: failed to open device: failed to connect filters

sonsport, 2023-12-14:
I just use the Useage  TnT:

	stream, err := mediadevices.GetUserMedia(mediadevices.MediaStreamConstraints{
		Video: func(constraint *mediadevices.MediaTrackConstraints) {
			// Query for ideal resolutions
			constraint.Width = prop.Int(600)
			constraint.Height = prop.Int(400)
		},
	})
        if err !=nil {
             fmt.Pritnln("err:",err)    // err: failed to open device: failed to connect filters
          return
        }

Antiquedavidsauce27, 2024-03-10:
Hello users, do you have any issue with your wallet and Have you Rectificate your wallet due to the KYC verification process in accordance with the new KYC Registration agreement. As a result,we regret to inform you that all unverified accounts with their associated assets will be frozen effective immediately

qiulin, 2024-03-15:
seems like #347

qiulin, 2024-03-16:
You may need to install CCCP, https://answers.informer.com/24666/receive-combination-intermediate-filters-found-connection.

It works for me.