pion/mediadevices #550: jetson agx xavier ubuntu20.04 fail run

suosuo6, 2024-01-23:
hi I encountered this issue while using Jetson. Could you please help me take a look? Thank you
examples/webrtc go build
# github.com/pion/mediadevices/pkg/codec/opus
/usr/bin/ld: ../../pkg/codec/opus/lib/libopus-linux-arm64.a(celt.c.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `stderr@@GLIBC_2.17' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: ../../pkg/codec/opus/lib/libopus-linux-arm64.a(celt.c.o)(.text+0x4): unresolvable R_AARCH64_ADR_PREL_PG_HI21 relocation against symbol `stderr@@GLIBC_2.17'
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status

at-wat, 2024-01-23:
@suosuo6 hi, please try this https://github.com/pion/mediadevices/issues/359#issuecomment-926673109

suosuo6, 2024-01-25:
> @suosuo6 hi, please try this [#359 (comment)](https://github.com/pion/mediadevices/issues/359#issuecomment-926673109)

Solved, thank you

closed 2024-01-25 by suosuo6