pion/mediadevices #622: vaapi does not support `card1` as for encoding

benbaldivia-maumai, 2025-04-15:
### Your environment.
  - Version: *v0.7.1*
  - Browser: *N/A*
  - Other Information - *https://github.com/pion/mediadevices/blob/20e8c5073579be9c885ea1ce232ce7de78bbdfa4/pkg/codec/vaapi/vp9.go#L198*

### What did you do?

The computers I am working with have `/dev/dri/card1` and not `/dev/dri/card0`.

### What did you expect?
For library to detect `card1` or to allow for specification of `/dev/dri/card1` via parameter or option when using `vaapi`

### What happened?
Error when failing to open `/dev/dri/card0`

3DRX, 2025-04-16:
@benbaldivia-maumai Hi, would you like to open a PR that add configurable hardware device to `codec/vaapi`? Similar to #621 :)

benbaldivia-maumai, 2025-04-16:
> @benbaldivia-maumai Hi, would you like to open a PR that add configurable hardware device to `codec/vaapi`? Similar to #621 :)

I definitely would be open to that. I can probably get around to it either later this week or on the weekend possibly. I will review stuff a bit first to see if I can adhere to how things are configured for pion.