pion/mediadevices #623: vaapi does not support VP9 Low Power encoding

benbaldivia-maumai, 2025-04-15:
### Your environment.
  - Version: *v0.7.1*
  - Browser: *N/A*
  - Other Information - *https://github.com/pion/mediadevices/blob/20e8c5073579be9c885ea1ce232ce7de78bbdfa4/pkg/codec/vaapi/vp9.go#L222-L229*

### What did you do?

The codec appears to be hard coded to use only `VAEntrypointEncSlice`. While my device appears to support encoding via`VAProfileVP9Profile0`, it only supports `VAEntrypointEncSliceLP`. This is not found as the entry point is distinct from `VAEntrypointEncSlice` and it is hard coded to only look for `VAEntrypointEncSlice`.

Here is `vainfo` output from my device:
```
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_14
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.3.1 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSliceLP
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSliceLP
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile1            :	VAEntrypointVLD
      VAProfileVP9Profile1            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile3            :	VAEntrypointVLD
      VAProfileVP9Profile3            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileHEVCMain422_10         :	VAEntrypointVLD
      VAProfileHEVCMain422_12         :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          :	VAEntrypointVLD
      VAProfileHEVCSccMain10          :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         :	VAEntrypointVLD
      VAProfileHEVCSccMain444         :	VAEntrypointEncSliceLP
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointEncSliceLP
```

### What did you expect?
When using `vaapi` with my device with libva installed and support for `VAProfileVP9Profile0` detected in the OS for it to work without an error.

### What happened?
I got `libva entrypoint not found` when attempting to use `vaapi` VP9 encoding.

3DRX, 2025-04-15:
Also, can you try using the ffmpeg based encoder in https://github.com/pion/mediadevices/pull/621 and see if that works for you?

benbaldivia-maumai, 2025-04-15:
@3DRX Sounds good. I will see about trying the ffmpeg encoder. As a note, `vaapi` should work, I expect, if it is used on a device with `VAEntrypointEncSlice` support. The devices I am using at the moment though, only support `VAEntrypointEncSliceLP` which is the Low Power implementation of the encoder (as far as I am aware). The code linked in the issue seems to be hard coded to look for only `VAEntrypointEncSlice`.

When I change the hard coded value to use `VAEntrypointEncSliceLP` though, I do end up getting an error after a few reads on a `vaEndPicture` call:
https://github.com/pion/mediadevices/blob/20e8c5073579be9c885ea1ce232ce7de78bbdfa4/pkg/codec/vaapi/vp9.go#L430

When I look at the error returned it simply says `internal encoding error` so it was hard for me to troubleshoot the issue from that point.

benbaldivia-maumai, 2025-04-16:
@3DRX It isn't exactly clear how to set this up. Either way, I was looking at the code and I don't actually see the vp9 codec used. I see vp8 and others. So in my case it doesn't look like it would work in it's current form. The reason I was looking to use VP9 was a combination of factors. The first being I want use VP8/9 for license and bitrate/quality reasons. VP8 is fast enough using CPU only but VP9 quality is much better for the same bitrate. The second is that I would like to utilize hardware acceleration because VP9 via CPU encoding results in a significant delay and the later intel CPUs I am working with support VP9 hardware encoding of VP9 ([intel docs](https://www.intel.com/content/www/us/en/docs/onevpl/developer-reference-media-intel-hardware/1-1/overview.html)). The hardware encoding of vp8 via intel iGPU ended after 10th gen and later GPUs have all seemed to support VP9.

3DRX, 2025-04-16:
@benbaldivia-maumai Hi, I've updated the ffmpeg based encoder module to support vp9_vaapi. One of the benefits using ffmpeg is that adding new codec support requires minimal code change :)

benbaldivia-maumai, 2025-04-16:
> @benbaldivia-maumai Hi, I've updated the ffmpeg based encoder module to support vp9_vaapi. One of the benefits using ffmpeg is that adding new codec support requires minimal code change :)

I will definitely try it out. 👍😺

closed 2025-09-30 by benbaldivia-maumai